AprilRobotics/apriltag #121: Corner Locations

myaldiz, 2021-01-07:
Hello,

I am trying to understand the sub-pixel corner output of the apriltags with respect to opencv Aruco. I wrote the below code:

[image]

`april_marker` has no additional padding, I use the marker with 1 pixel black border (as is), upsample with the nearest neighbor setting and place it inside the image. So it is perfect placement. I used python wrapper in the recognize code, ordered and printed the output of the key `lb-rb-rt-lt` from python wrapper. 

From the values, I think apriltags assumes (0.5, 0.5) offset for the pixel center, as opposed to opencv having no offset. I'd like to ask whether my intuition from the numbers I get is correct. And is there a reason why it does not provide more accurate  locations for the corners, given that it is most perfect an image can get?

Thanks!

mkrogius, 2021-03-17:
I believe apriltags does assume 0.5, 0.5 for the pixel center.

It is possible that your bilinear interpolation has effectively blurred the image of the tag, which would reduce the accuracy of the corner locations

myaldiz, 2021-03-17:
I used perfect placement (with nearest neighbor upsampling), so there is not bilinear interpolation. Despite that, I confirmed above results (apriltag having 0.5 offset, and aruco has none) with rendered experimental setting.

closed 2021-03-17 by myaldiz