AprilRobotics/apriltag #125: import apriltag errors

n-gineer, 2021-01-20:
I'm using the basic Python example shown in https://github.com/AprilRobotics/apriltag/blob/master/apriltag_detect.docstring.

I created a new file and named it apriltag3_sample.py

when trying to run with 
`python apriltag3_sample.py`
it returned:
`Traceback (most recent call last):
  File "apriltag3_sample.py", line 7, in <module>
    detector = apriltag("tag36h11")
TypeError: 'module' object is not callable`

mkrogius, 2021-01-21:
This is odd. What this means is that the symbol 'apriltag' at this point in your code is a module not a class. But it should be a class because the script imports it as 'from apriltag import apriltag'.

Can you run the following python script and report the results? It will mostly likely give an error, that's ok, just report the error too.

```
import apriltag
print('Testing apriltag')
print(type(apriltag))
print(dir(apriltag))
print(apriltag.__file__)

from apriltag import apriltag
print('Testing apriltag.apriltag')
print(type(apriltag))
print(dir(apriltag))
print(apriltag.__file__)

print('Testing apriltag.apriltag.apriltag')
print(type(apriltag.apriltag))
print(dir(apriltag.apriltag))
print(apriltag.apriltag.__file__)
```

mkrogius, 2021-01-21:
That is odd. What this means is that the name 'apriltag' is a module instead of a class at that line in your code, but it should be a class because of the import line 'from apriltag import apriltag'.

Can you run the following python script and report the results?

n-gineer, 2021-01-22:
ran and returned:
`Testing apriltag
<class 'module'>
['__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__']
None
Testing apriltag.apriltag
<class 'module'>
['__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'apriltag']
/home/pi/.virtualenvs/cv/lib/python3.7/site-packages/apriltag/apriltag.cpython-37m-arm-linux-gnueabihf.so
Testing apriltag.apriltag.apriltag
<class 'type'>
['__class__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'detect']
Traceback (most recent call last):
  File "apriltag/troubleshoot_125.py", line 16, in <module>
    print(apriltag.apriltag.__file__)
AttributeError: type object 'apriltag' has no attribute '__file__'`

mkrogius, 2021-01-22:
Thanks!

It looks like the problem is that your apriltag.cpython-37m-arm-linux-gnueabihf.so file got installed in a different location than normal. It ended up in python3.7/site-packages/apriltag/apriltag.cpython-37m-arm-linux-gnueabihf.so instead of python3.7/site-packages/apriltag.cpython-37m-arm-linux-gnueabihf.so

The easy fix is just to move the apriltag.cpython-37m-arm-linux-gnueabihf.so file up one level. The better fix is to figure out why the install location from the makefile is incorrect on your machine.

n-gineer, 2021-01-24:
> Thanks!
>
> It looks like the problem is that your apriltag.cpython-37m-arm-linux-gnueabihf.so file got installed in a different location than normal. It ended up in [...]

I wasn't completely clear on the "Install" section of the ReadMe, but was this:

"
If you have CMake installed or it is not difficult to install, then do:
```
$ cmake .
$ sudo make install 
```
"
supposed to be done in "python3.7/site-packages/" or inside of "python3.7/site-packages/apriltag/"?

mkrogius, 2021-01-25:
No, both of these should be done from wherever you cloned the git repo.

n-gineer, 2021-01-30:
> No, both of these should be done from wherever you cloned the git repo.
I had cloned into 
`python3.7/site-packages/apriltag/`
and ran cmake & make there. Would this have caused the problem?

mkrogius, 2021-02-01:
Yes, that is the problem. It will work properly if you clone the git repo pretty much anywhere else

closed 2021-02-01 by mkrogius

brmarkus, 2026-02-28:
> import cv2 import apriltag import numpy as np import serial import time
> 
> detector = apriltag.apriltag( family='tag36h11', threads=4, maxhamming=1, decimate=2.0, blur=0.0, refine_edges=True, debug=False )
> 
> when i run the following program which i written it shows an error like this
> 
> Traceback (most recent call last): File "/home/lambda3/My_Workspace/Office_Work/AGV_Projects/Python_Projects/53am23ed_Motor_gets_stopped_based_on_Logitech_camera/captures_tagid_sends_cmd_to_pi.py", line 7, in detector = apriltag.apriltag( ^^^^^^^^^^^^^^^^^ AttributeError: module 'apriltag' has no attribute 'apriltag'

Have a look into the example from the documentation under
https://github.com/AprilRobotics/apriltag?tab=readme-ov-file#python

```
import cv2
import numpy as np
from apriltag import apriltag

imagepath = 'test.jpg'
image = cv2.imread(imagepath, cv2.IMREAD_GRAYSCALE)
detector = apriltag("tagStandard41h12")

detections = detector.detect(image)
```

Would that work for you @harishkumar0004 ?