AprilRobotics/apriltag #128: Problem with find marker

sars-ivan-by, 2021-01-27:
We are using your AprilTag library. Our beginnings were connected with AprilTag 2, but now we
have started porting our application to AprilTag 3. We got some bugs that weren't with the previous
version: “WRN: Matrix is singular.”, “Error, more than one new minimum found.”, “malloc():
mismatching next->prev_size (unsorted)”. Also I have question about criteria for error at a distance from camera to marker and degrees of rotation markers. Also we have problem "too many border/line", desirable  solve this problem without any quad_contours. If we use divide frame to solve it, we can't find some parameters: distance and rotation`

detector=apriltags3.Detector(quad_decimate=1,decode_sharpening=1)
detections = detector.detect(img, estimate_tag_pose=False,camera_params=[fx,fy,cx,cy],tag_size=60)` Also img can be after divided

mkrogius, 2021-01-28:
Please post one "github issue" per issue you are having. Also, please include enough information with each issue so that I can debug it, e.g. an image and apriltag command that reproduces each bug.

closed 2021-01-28 by mkrogius

sars-ivan-by, 2021-01-29:
Thank you, i post new "github issue" per issue