AprilRobotics/apriltag #130: problem with find markers

sars-ivan-by, 2021-01-29:
i get error : Error, more than one new minimum found.
malloc(): mismatching next‑>prev_size (unsorted)
[image]
`detector = apriltags3.Detector()
    detector_2 = apriltags3.Detector(families='tag25h9')
    fx=mtx[0][0]
    fy=mtx[1][1]
    cx=mtx[0][2]
    cy=mtx[1][2]
    detections = detector.detect(img_cv_gray,estimate_tag_pose=True,camera_params=[fx,fy,cx,cy],tag_size=90)
    detections_2 = detector_2.detect(img_cv_gray,estimate_tag_pose=True,camera_params=[fx,fy,cx,cy],tag_size=90)`

mkrogius, 2021-02-01:
It looks like you aren't using the python wrapper from this library? Whose apriltag code are you using?

sars-ivan-by, 2021-02-02:
I use this https://github.com/duckietown/apriltags3-py code. What python wrapper did you mean?

nzjrs, 2021-02-02:
> I use this https://github.com/duckietown/apriltags3-py code. What python wrapper did you mean?

the python wrapper in this repository...

sars-ivan-by, 2021-02-02:
@nzjrs can you link on python wrapper?

nzjrs, 2021-02-02:
> @nzjrs can you link on python wrapper?

https://github.com/AprilRobotics/apriltag#install (read the part about python)

mkrogius, 2021-02-13:
I'm closing this issue since it doesn't happen when using the python wrapper from this repo.

closed 2021-02-13 by mkrogius