AprilRobotics/apriltag #131: criteria for error: at a distance

sars-ivan-by, 2021-01-29:
I use Apriltag2, but when I tried to use Apriltag3, I had seen, that Apriltag2 is better than Apriltag3 on same camera params. I did something wrong or it will be better to use Apriltag2?

mkrogius, 2021-02-01:
More details are needed. In what ways is AprilTag 2 vs 3 better? In theory, AprilTag 3 should be an improvement in every way

sars-ivan-by, 2021-02-02:
when I calibrated my iphone and check results about distance from me to markers on same video, I find that Apriltag2 better. And when I divided frame I can't find rotation matrix on Apriltag3.

mkrogius, 2021-02-13:
In both the other issues you posted, you weren't using the python wrapper from this repo. I'm going to go ahead and close this issue too since I'm assuming that this is also the duckietown python wrapper.

closed 2021-02-13 by mkrogius