AprilRobotics/apriltag #136: Scripts for Instant Tag Compiles

dajoke, 2021-03-05:
The tag files can take incredibly long to compile due to their structure. Here are a few scripts that reformat the tag*.c files so they compile ~instantly.  Run "csh tagfix.csh tagStandard52h13.c" for example.  

The scripts preserve the existing structure of putting the codes data into malloc'd storage, though if it never changes then it could just be left in the static array and only the pointer put into the struct. Presumably the tag file generator could be made to do this directly; I haven't tracked that down.
[tagfix.zip](https://github.com/AprilRobotics/apriltag/files/6092721/tagfix.zip)

dajoke, 2021-03-05:
FYI, here's an excerpt of the transform the scripts apply to supercharge compile time, here to tag16h5.c :

```
#include <stdlib.h>
#include "tag16h5.h"

static uint64_t codedata[30] = {
0x00000000000027c8UL,
0x00000000000031b6UL,
0x0000000000003859UL,
0x000000000000569cUL,
0x0000000000006c76UL,
0x0000000000007ddbUL,
0x000000000000af09UL,
0x000000000000f5a1UL,
0x000000000000fb8bUL,
0x0000000000001cb9UL,
0x00000000000028caUL,
0x000000000000e8dcUL,
0x0000000000001426UL,
0x0000000000005770UL,
0x0000000000009253UL,
0x000000000000b702UL,
0x000000000000063aUL,
0x0000000000008f34UL,
0x000000000000b4c0UL,
0x00000000000051ecUL,
0x000000000000e6f0UL,
0x0000000000005fa4UL,
0x000000000000dd43UL,
0x0000000000001aaaUL,
0x000000000000e62fUL,
0x0000000000006dbcUL,
0x000000000000b6ebUL,
0x000000000000de10UL,
0x000000000000154dUL,
0x000000000000b57aUL,
};

apriltag_family_t *tag16h5_create()
{
   apriltag_family_t *tf = calloc(1, sizeof(apriltag_family_t));
   tf->name = strdup("tag16h5");
   tf->h = 5;
   tf->ncodes = 30;
   tf->codes = calloc(30, sizeof(uint64_t));

   int i;
   for (i = 0; i < 30; i++)
      tf->codes[i] = codedata[i];

   tf->nbits = 16;
   tf->bit_x = calloc(16, sizeof(uint32_t));
   tf->bit_y = calloc(16, sizeof(uint32_t));
   tf->bit_x[0] = 1;
   tf->bit_y[0] = 1;
   tf->bit_x[1] = 2;
   tf->bit_y[1] = 1;
   ...
```

mkrogius, 2021-03-05:
Have you tested this on the larger tag families? I tried a number of variations along this line but the compiler didn't like it when the array size gets too large (I can't remember if I tried multiple static arrays, I suspect I did try it and the compiler still didn't like it). In the end I settled for fiddling with the compiler flags to try to get it to spend less time on the tag files. It takes about a minute or so for a clean build on my machine.

Out of curiousity, what compiler version are you using and how long does it take for that first build? Also, are you using cmake or are you using the handwritten makefile? https://github.com/AprilRobotics/apriltag#install

FYI, the c files are autogenerated using: https://github.com/AprilRobotics/apriltag-generation/blob/master/src/april/tag/TagToC.java
It may be easier to change how the files are generated there.

dajoke, 2021-03-05:
I ran it on all the families and compiled them lickety-split. The compile was taking like an hour for the big families originally (my default settings are for full optimization for everything, so that surely didn't help). This was on a M1 mac at the moment, so it's using clang. I'll see what happens on my linux machine in a little bit since you ask.  It compiles PDQ on Windows no problem (MSVC 2015).  I'm currently working on stubbing pthreads out to windows primitives. 

I don't use cmake or the existing makefiles, I use a simple flattening scheme that makes it easier since I need to maintain 4 different platforms (and multiple libraries on each). Makefiles etc are good when you're developing, but not necessary for "production" libraries, there simplicity and lack of dependencies is better.

dajoke, 2021-03-05:
gcc 8.3.1 serially compiles the entire library and all the tag files in <7 sec on an older quad-core linux machine.

mkrogius, 2021-03-08:
You were right, this change works for even the larger tag families, thanks for the good idea!

The change to the code generation is here: https://github.com/AprilRobotics/apriltag-generation/commit/32e7caa61d4776d8e24e70a2484a4590d0f1afbf

closed 2021-03-08 by mkrogius