AprilRobotics/apriltag #138: How can I obtain 4 corner coordinates?

liuyang8218, 2021-03-19:
Hi there,

    As we know each tag can only provide X-Y-Z information of the center point, I'm wondering can I also get those information of 4 corner points？
    I noticed that through Point(p[0][0],p[0][1]) can only get x,y pixel coordinates. 

    Please let me know if you have any idea.

    Thank you!
    Best regards

mkrogius, 2021-03-19:
The pose estimate gives you not just the center of the tag, but also the rotation of the tag. From this you can calculate the position of each corner

closed 2021-03-19 by mkrogius

liuyang8218, 2021-03-22:
Thank you Max,

The problem is I need precise position of each corner, the rotation of tag exist a huge noise even if I  change the tag size. Is there another way to solve this?

liuyang8218, 2021-03-22:
> The pose estimate gives you not just the center of the tag, but also the rotation of the tag. From this you can calculate the position of each corner

Thank you Max,

The problem is I need precise position of each corner, the rotation of tag exist a huge noise even if I change the tag size. Is there another way to solve this?