AprilRobotics/apriltag #141: detect tags with encroaching black squares

berndpfrommer, 2021-04-08:
Calibration boards often have encroaching black squares like here:
https://github.com/ethz-asl/kalibr/wiki/calibration-targets
These squares are useful because they allow for corner refinement to reduce corner location errors.

Since the UMich Apriltag detector does not work with such tags, people typically use Michael Kaess' implementation for that:
https://bitbucket.org/kaess/apriltags/src/master/
But that one is no longer actively maintained.

Question: is there a way to get the UMich Apriltag3 detector to work in such situations? Where exactly in the detection algorithm does it fail? Any suggestions on how this could be improved?
Thanks!

mkrogius, 2021-04-08:
The switch from AprilTag 1 -> AprilTag 2 changed how the quadrilateral detection works. What the more modern detector looks for is a contiguous white region that neighbors/encloses/is enclosed by, a contiguous black region. Those extra squares are splitting the white border into multiple disconnected elements, so the tag detection will fail at a pretty fundamental level.

AprilTag 3 has a solution for this problem, which I haven't seen anyone use yet. We now allow tag families which have the black border on the outside, and the white border on the inside. So you could make a new sort of checkerboard, which has alternating tags from two different families, the classic 36h11 family, and a new 36h11 family with the borders reversed. This would be fully detectable by the AprilTag 3 detector.

berndpfrommer, 2021-04-09:
Alright then, no easy fix.
Making a calibration board as you described is indeed an interesting idea that I may try out. The drawbacks are that I'd have to upgrade my calibration boards and that Kalibr, which is no longer actively maintained, will not work with the new boards. Still thanks for your prompt reply!

closed 2021-04-09 by berndpfrommer

reprojection-calibration, 2026-05-16:
Link to a small wiki article with some more details: https://github.com/reprojection-calibration/reprojection/wiki/Aprilgrid3