AprilRobotics/apriltag #142: apriltag detect always return nothing

WingEternal, 2021-04-28:
Hello! I modified opencv_demo.cc and input an image with tagStandard41h12 for testing, but apriltag_detector_detect() always return an empty z_array. Are there any other parameters that need to be given?
My code:
    
    Mat gray_image = cv::imread("./test1_.png");
    image_u8_t im = { .width = gray_image.cols, .height = gray_image.rows, .stride = gray_image.cols, .buf = gray_image.data};
    std::cout<<gray_image.cols<<" "<<gray_image.rows<<std::endl;
    // Initialize tag detector with options
    apriltag_family_t *tf = tagStandard41h12_create();

    apriltag_detector_t *td = apriltag_detector_create();
    apriltag_detector_add_family(td, tf);
    td->quad_decimate = 1.0f;
    td->quad_sigma = 0.0f;
    td->nthreads = 2;
    td->debug = 0;
    td->refine_edges = 1;

    zarray_t *detections = apriltag_detector_detect(td, &im);
    std::cout<<detections->size<<std::endl;
    // Draw detection outlines
    for (int i = 0; i < zarray_size(detections); i++) {
        apriltag_detection_t *det;
        zarray_get(detections, i, &det);
        line(gray_image, Point(det->p[0][0], det->p[0][1]),
                    Point(det->p[1][0], det->p[1][1]),
                    Scalar(0, 0xff, 0), 2);
        line(gray_image, Point(det->p[0][0], det->p[0][1]),
                    Point(det->p[3][0], det->p[3][1]),
                    Scalar(0, 0, 0xff), 2);
        line(gray_image, Point(det->p[1][0], det->p[1][1]),
                    Point(det->p[2][0], det->p[2][1]),
                    Scalar(0xff, 0, 0), 2);
        line(gray_image, Point(det->p[2][0], det->p[2][1]),
                    Point(det->p[3][0], det->p[3][1]),
                    Scalar(0xff, 0, 0), 2);

        stringstream ss;
        ss << det->id;
        String text = ss.str();
        int fontface = FONT_HERSHEY_SCRIPT_SIMPLEX;
        double fontscale = 1.0;
        int baseline;
        Size textsize = getTextSize(text, fontface, fontscale, 2,
                                        &baseline);
        putText(gray_image, text, Point(det->c[0]-textsize.width/2,
                                    det->c[1]+textsize.height/2),
                fontface, fontscale, Scalar(0xff, 0x99, 0), 2);
    }
    apriltag_detections_destroy(detections);

    imshow("Tag Detections", gray_image);
    waitKey();

    apriltag_detector_destroy(td);
    tagStandard41h12_destroy(tf);`
test1.png ( gray scale, 1825 x 1020 ):

[image]

mkrogius, 2021-05-07:
If I convert the image to .pnm and use apriltag_demo as is, it successfully detects the tag. Maybe this is something to do with how you are reading the image?

You can try setting debug = 1 and looking at the debug images produced to see what is happening as this image is processed

WingEternal, 2021-05-12:
Thinks for help!
I found that the problem is caused by cv::imread(), which reads images as color images by default. The problem is solved by changing
`Mat gray_image = cv::imread("./test1_.png");`
to
`Mat gray_image = cv::imread("./test1_.png", IMREAD_GRAYSCALE);`

closed 2021-05-12 by WingEternal