AprilRobotics/apriltag #144: "Undefined reference" error in ROS2 using OpenCV

unia95, 2021-05-04:
Hi everybody, I'm new to AprilTags and I was trying to include them in my project. 
I basically have a ROS2 node that reads a flow from a camera and, if present, detects a tag, finds its number and computes its position. This is done in C++ language and using OpenCV 4 with CUDA for processing the images.
So I downloaded and built the library according to the guidelines, and I included it in my c++ code, as follows:

//all the other included libraries
extern "C" {
#include <apriltag/apriltag.h>
#include <apriltag/tag36h11.h>
#include <apriltag/tag25h9.h>
#include <apriltag/tag16h5.h>
#include <apriltag/tagCircle21h7.h>
#include <apriltag/tagCircle49h12.h>
#include <apriltag/tagCustom48h12.h>
#include <apriltag/tagStandard41h12.h>
#include <apriltag/tagStandard52h13.h> 
#include <apriltag/common/getopt.h>
}

but when I try to build the code, I get all these errors:

CMakeFiles/airsim_img_listener.dir/src/show_img.cpp.o: In function `apriltag_detector_add_family':
show_img.cpp:(.text+0x336): undefined reference to `apriltag_detector_add_family_bits'
CMakeFiles/airsim_img_listener.dir/src/show_img.cpp.o: In function `ImgViewSubscriber::ImgViewSubscriber()':
show_img.cpp:(.text._ZN17ImgViewSubscriberC2Ev[_ZN17ImgViewSubscriberC5Ev]+0xa93): undefined reference to `tagStandard41h12_create'
show_img.cpp:(.text._ZN17ImgViewSubscriberC2Ev[_ZN17ImgViewSubscriberC5Ev]+0xa9f): undefined reference to `apriltag_detector_create'
CMakeFiles/airsim_img_listener.dir/src/show_img.cpp.o: In function `ImgViewSubscriber::front_callback(std::shared_ptr<sensor_msgs::msg::Image_<std::allocator<void> > >) const':
show_img.cpp:(.text._ZNK17ImgViewSubscriber14front_callbackESt10shared_ptrIN11sensor_msgs3msg6Image_ISaIvEEEE[_ZNK17ImgViewSubscriber14front_callbackESt10shared_ptrIN11sensor_msgs3msg6Image_ISaIvEEEE]+0x22a): undefined reference to `apriltag_detector_detect'
show_img.cpp:(.text._ZNK17ImgViewSubscriber14front_callbackESt10shared_ptrIN11sensor_msgs3msg6Image_ISaIvEEEE[_ZNK17ImgViewSubscriber14front_callbackESt10shared_ptrIN11sensor_msgs3msg6Image_ISaIvEEEE]+0x875): undefined reference to `apriltag_detections_destroy'
show_img.cpp:(.text._ZNK17ImgViewSubscriber14front_callbackESt10shared_ptrIN11sensor_msgs3msg6Image_ISaIvEEEE[_ZNK17ImgViewSubscriber14front_callbackESt10shared_ptrIN11sensor_msgs3msg6Image_ISaIvEEEE]+0x925): undefined reference to `apriltag_detector_destroy'
show_img.cpp:(.text._ZNK17ImgViewSubscriber14front_callbackESt10shared_ptrIN11sensor_msgs3msg6Image_ISaIvEEEE[_ZNK17ImgViewSubscriber14front_callbackESt10shared_ptrIN11sensor_msgs3msg6Image_ISaIvEEEE]+0x934): undefined reference to `tagStandard41h12_destroy'
collect2: error: ld returned 1 exit status
make[2]: *** [airsim_img_listener] Error 1
make[1]: *** [CMakeFiles/airsim_img_listener.dir/all] Error 2
make: *** [all] Error 2

I've already included the "find_package(apriltag REQUIRED)" line in my project's CMakeLists.txt file and the "<depend>apriltag</depend>" line in my package.xml file.

Any hint to help me to solve that issue? It's very frustrating and nobody seems to have experienced that before.
Thanks in advance!

wxmerkt, 2021-05-04:
This appears to be a linking error. Are you linking to the library / target? Can you post the excerpt of the CMakeLists where you configure (find_package), create the library/executable etc.?

unia95, 2021-05-04:
Thanks for your kind help. I'm posting the whole CMakeLists.txt file, just to be sure:

cmake_minimum_required(VERSION 3.5)
project(major_tom_cpp)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(OpenCV 4.5.2 REQUIRED) #3.2.0 REQUIRED)
find_package(cv_bridge REQUIRED)
find_package(apriltag REQUIRED)
find_package(tf2_msgs REQUIRED)

add_executable(airsim_img_listener src/show_img.cpp)
ament_target_dependencies(airsim_img_listener rclcpp std_msgs sensor_msgs OpenCV cv_bridge apriltag)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  ament_lint_auto_find_test_dependencies()
endif()

install(TARGETS
  airsim_img_listener
  DESTINATION lib/${PROJECT_NAME})

ament_package()

unia95, 2021-05-05:
I solved my problem by following this guide:
https://github.com/amy-tabb/apriltag-cmake-example
and now my code works, it detects the marks as expected.
I'm closing the issue :D

closed 2021-05-05 by unia95