AprilRobotics/apriltag #147: UDP receive from iOS application

jcnwholland, 2021-06-10:
Hi,

First of all, many thank for your great work. I'm not an electric engineer, but a medical doctor with a hobby interest in automated tracking and navigating.

I'm running into a problem trying to receive the UPD stream from the IOS deceive, mainly the javascript for decoding. I've tried many methods but in windows I cannot import inside the node or module.

"Cannot use import statement inside the Node.js REPL, alternatively use dynamic import"
 
I can grab the stream with a simple UDP tool in windows, getting this in the data stream:
in ASCII
`" RX 284; 192.168.178.150:58411> APRILTAG"`  with some unrecognized character
```
In HEX:
"41 50 52 49 4C 54 41 47 0 1 0 2 0 0 0 50 0 5 FFFD 66 FFFD 4D 7D FFFD 0 0 0 E 0 0 0 0 0 0 2 4B 44 30 53 5B 44 40 77 FFFD 44 33 47 FFFD 44 48 FFFD 7D 44 34 FFFD 5B 44 3D 52 FFFD 44 2D 74 3E 44 38 6F 45 44 2B FFFD 24 44 43 FFFD FFFD 40 FFFD FFFD FFFD 41 0 FFFD "
```

Alternatively i've tried grabbing it in MATLAB, but cant get the javascript to run, and with instrument manager I get nothing more then "Apriltag" over the UDP stream as data. 

So my question, how do I use the script in a windows or MATLAB environment? So it can store tag ID and position?  Or how do i get by the java import limitation? Or is there a simpler method that I as newbie are missing?
Have a dual boot, so Ubuntu is also an option.

Thank you, Gr Joep
JCN WIllemsen MD PhD
Holland

jcnwholland, 2021-06-10:
The script is from apriltag website;

`import java.io.*;
import java.net.*;

public  class AprilTagReceive
{
    public static void main(String args[]) throws IOException
    {
        DatagramSocket sock = new DatagramSocket(7709);

        byte buf[] = new byte[65536];
        DatagramPacket packet = new DatagramPacket(buf, buf.length);

        while (true) {
            sock.receive(packet);

            DataInputStream ins = new DataInputStream(new ByteArrayInputStream(buf));
            int MAGIC1 = ins.readInt();
            int MAGIC2 = ins.readInt();
            assert(MAGIC1 == 0x41505249);
            assert(MAGIC2 == 0x4c544147);

            int version = ins.readInt();
            assert(version == 0x00010001);

            int ndets = ins.readInt();
            long utime = (((long)ins.readInt())<<32) + ins.readInt();

            System.out.printf("%13.6f: %d tags\n", utime/1000000.0, ndets);
            for (int i = 0; i < ndets; i++) {
                int id = ins.readInt();
                int hamming = ins.readInt();
                float goodness = ins.readFloat();

                double c[] = new double[2];
                for (int j = 0; j < 2; j++)
                    c[j] = ins.readFloat();

                double p[][] = new double[4][2];
                for (int j = 0; j < 4; j++) {
                    p[j][0] = ins.readFloat();
                    p[j][1] = ins.readFloat();
                }

                double H[] = new double[9];
                for (int j = 0; j < 9; j++)
                    H[j] = ins.readFloat();

                System.out.printf(" id=%3d, err=%d at (%8.2f,%8.2f)\n",
                                  id, hamming, c[0], c[1]);
            }
        }
    }
}
`
The udp data is from the ios app..

mkrogius, 2021-06-10:
Ah I see, ok there's a few things going on here.

1) It is normal that the generic simple UDP tool would see random characters, it doesn't know how to decode the data we are sending. But, it is good that it sees something, that means the udp packets are being received correctly.

2) The AprilTagReceive.java file is a file written in the java programming language ([not javascript](https://www.thesoftwareguild.com/faq/difference-between-java-and-javascript/)). I would recommend using a Java IDE such as IntelliJ or Eclipse to run this file.

3) It looks like the AprilTagReceive.java file was not updated for AprilTag3. It looks like the packet now has slightly different information in it, which you can see in the udp transmit code here: https://github.com/edwinolson/apriltags-ios/blob/master/AprilTag3/AppDelegate.m#L1605

I don't have an iOS device so I can't be sure that my changes will work, but I attempted to update AprilTagReceive.java here: https://gist.github.com/mkrogius/8f03e335d07cd8bd27e51b8e4d46cf1f

jcnwholland, 2021-06-10:
@mkrogius You are a legend!! Did a quick test before bedtime, and i'm getting data! Will run some more tests tomorrow and i'll let you know how it handels.

Thanks a bunch ;-) Gr Joep

closed 2022-03-10 by wxmerkt