AprilRobotics/apriltag #148: Make Error | Nvidia Jetson Nano

maharshi114, 2021-06-14:
Command

`sudo make install `

Error : 

```
[100%] Linking CXX executable opencv_demo
/home/nvidia/yes/envs/jupyter/lib/libopencv_videoio.so.4.5.2: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
collect2: error: ld returned 1 exit status
CMakeFiles/opencv_demo.dir/build.make:149: recipe for target 'opencv_demo' failed
make[2]: *** [opencv_demo] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/opencv_demo.dir/all' failed
make[1]: *** [CMakeFiles/opencv_demo.dir/all] Error 2
```

System Information : 

OS: Ubuntu 18.04.5 LTS (GNU/Linux 4.9.201-tegra aarch64)
Hardware: Jetson Nano 
Installation Environment: Conda Jupyter
OpenCV Version: 4.5.2
Python Version: 3.6.13

mkrogius, 2021-06-14:
It looks like the issue is something to do with mismatched c++ stdlib versions

Can you delete line 150 of the CMakeLists.txt? It should be this line: "set_target_properties(opencv_demo PROPERTIES CXX_STANDARD 11)"

After deleting that line then do a 'make clean' and then try building again

maharshi114, 2021-06-15:
Thank you, 

According to your suggestion, I have implemented the same in CMakeLists.txt. 
But the error is still the same.

[100%] Linking CXX executable opencv_demo
/home/nvidia/yes/envs/jupyter/lib/libopencv_videoio.so.4.5.2: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
collect2: error: ld returned 1 exit status
CMakeFiles/opencv_demo.dir/build.make:149: recipe for target 'opencv_demo' failed
make[2]: *** [opencv_demo] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/opencv_demo.dir/all' failed
make[1]: *** [CMakeFiles/opencv_demo.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

maharshi114, 2021-06-15:
I found a problem in the environment setup. 

Finally i can able to make a library in the environment.

closed 2021-06-15 by maharshi114