AprilRobotics/apriltag #149: Choosing a Tag Family for maximum detection range and robustness

Horus26, 2021-06-15:
Hi,

which tag family should i choose when robust detection and detection for long ranges (>20m) are priorities, but only less than 20 tags are required?
I get that a larger Hamming Distance leads to a more robust detection and reduces the amount of tags in a family and that less data bits lead to a larger detection range, but also more false positives (less robust).
What is a good tradeoff between hamming distance and data bits concerning long range detection and robustness. (Maybe for creating a custom tag family)

Thanks in advance.

maharshi114, 2021-06-15:
I prefer 36h11, Reference : [https://april.eecs.umich.edu/media/pdfs/krogius2019iros.pdf](https://april.eecs.umich.edu/media/pdfs/krogius2019iros.pdf)

Horus26, 2021-06-15:
> 
> 
> I prefer 36h11, Reference : https://april.eecs.umich.edu/media/pdfs/krogius2019iros.pdf

Thanks for your answer.
The tag36h11 family is believed to perform better in this experiment setup due to the given dataset (large white background) as stated in Section 4.C. of https://april.eecs.umich.edu/media/pdfs/krogius2019iros.pdf.
As stated in section 4.A the tagStandard41h12 family should always be detected from larger distances than the tag36h11 family, when printed the same size. Therefore the old families are not considered.

mkrogius, 2021-06-15:
My default recommendation is tagStandard41h12, but if you want longer ranges and don't need many tags then you could go one size smaller and make a tagStandard32h11, but this will give you only about ~11% improvement in range.

I would also mention that the other good ways to increase detection range are to use a higher resolution camera or a lens with greater magnification, or to print the tag larger.

Horus26, 2021-06-15:
> My default recommendation is tagStandard41h12, but if you want longer ranges and don't need many tags then you could go one size smaller and make a tagStandard32h11, but this will give you only [...]

Thanks for the recommendation, i will definitely give it a shot.  I am limited by a certain camera, therefore changing the tag family is the only option. 
Could you elaborate on how you calculate or estimate the 11%?

mkrogius, 2021-06-15:
You could maybe still print the tag itself larger?

I estimated the range increase as follows. The tagStandard41h12 family is 9 cells wide while tagStandard32h11 would be 8 cells wide. This means that a cell from tagStandard32h11 would be imaged at the same size as one from the tagStandard41h12 family if it were at 9/8 or 1.125 times the distance (I should have said 12.5% above, but I didn't bother thinking enough).

Horus26, 2021-06-15:
ok, thanks for the elaboration. I close this issue now. You created a really nice system. Thanks for your awesome work!!

closed 2021-06-15 by Horus26