AprilRobotics/apriltag #150: Lower FPS NVIDIA Jetson Nano

maharshi114, 2021-06-15:
One of the python wrappers of the April tag library that I am using for this benchmark test. [https://github.com/duckietown/lib-dt-apriltags](https://github.com/duckietown/lib-dt-apriltags)

Intel NUC i3 CPU Results: 
1. 30 FPS Apriltag not in FOV
2. 30 FPS Apriltag in FOV

Nvidia Jetson nano:
1. 22 FPS Apriltag not in FOV
2. 5 FPS Apriltag in FOV

the code is the same.

mkrogius, 2021-06-15:
That is very weird that the FPS drops when the AprilTag is in frame. This implies that the tag decode process is taking most of the extra time, which shouldn't happen.

As far as I know the code does not have this issue on other ARM cpus, so it is possible this is something specific to the NVIDIA Jetson. As I do not have a jetson, I cannot debug this myself but I am happy to provide some amount of guidance should you want to debug this yourself.

Another possibility is the lib-dt-apriltags maybe is not building correctly for ARM? You could test this easily by building the official library from source on the jetson.

maharshi114, 2021-06-17:
Yes, I want to debug the code for Nvidia Jetson nano.

mkrogius, 2021-06-17:
First try building the library from source yourself on the jetson nano, to make sure it is build correctly for ARM

maharshi114, 2021-06-21:
How can I built/cross-compile the Apriltag library? can you share any basic steps, I will follow, or any other resources that I can follow to build this Apriltag library for Nvidia Jetson Nano?

mkrogius, 2021-06-21:
Can you please follow the [normal build+install instructions](https://github.com/AprilRobotics/apriltag#install),  but do them onboard the jetson?

maharshi114, 2021-06-22:
yes, I have already built this and tested it on Nvidia Jetson Nano.

mkrogius, 2021-06-22:
Then it is pretty surprising that this issue is still happening. The next thing to do is to produce the debug output ("-debug" on apriltag_demo) on both systems and compare the results to see if there are any noticeable differences there. Both sets of debug images will probably look the same but its best to start there before doing anything more involved.

After that, the next step will be to use a profiling tool such as perf to try to figure out which function is slow on the jetson. This will require building the code with debug symbols and then following a tutorial such as this one: http://sandsoftwaresound.net/perf/perf-tutorial-hot-spots/

jrepp, 2022-02-23:
It might be useful to get a trace from apriltag_demo for a single frame on both systems.

We are going to be running tests with apriltag3 on jetson hardware. If I discover this problem I will be able to fix it.

maharshi114, 2022-02-26:
@jrepp sure, it will helpful for opensource development.