AprilRobotics/apriltag #152: Using custom generated tag family in python

Horus26, 2021-06-17:
Hi again,

I generated the family tagStandard32h11 following the instructions in https://github.com/AprilRobotics/apriltag-generation.
I reinstalled Apriltag3 afterwards and in the debug output while installing i could find the new family. Therefore i am assuming a correct installation.
When trying to use the new family in python i get the error: Unrecognized tag family name: 'tagStandard32h11'. Families I know about:
  tag36h11
  tag25h9
  tag16h5
  tagCircle21h7
  tagCircle49h12
  tagStandard41h12
  tagStandard52h13
  tagCustom48h12

I used "tagStandard32h11" as family parameter for the detector. 
Therefore my question is: 
Is the functionality of adding and using new families even implemented for Python? (It seems like the possible tag families are hard coded in the python wrapper)

mkrogius, 2021-06-17:
Yes, they are hardcoded in the python wrapper. You can add your new family here: https://github.com/AprilRobotics/apriltag/blob/master/apriltag_pywrap.c#L28

Horus26, 2021-06-18:
I added the tag family as you can see in the following code and did the installation procedure (cmake . and sudo make install as above)

#include "apriltag.h"
#include "tag36h11.h"
#include "tag25h9.h"
#include "tag16h5.h"
#include "tagCircle21h7.h"
#include "tagCircle49h12.h"
#include "tagCustom48h12.h"
#include "tagStandard41h12.h"
#include "tagStandard52h13.h"
#include "tagStandard32h11.h"

#define SUPPORTED_TAG_FAMILIES(_)           \
    _(tag36h11)                             \
    _(tag25h9)                              \
    _(tag16h5)                              \
    _(tagCircle21h7)                        \
    _(tagCircle49h12)                       \
    _(tagStandard41h12)                     \
    _(tagStandard52h13)                     \
    _(tagStandard32h11)		     \
    _(tagCustom48h12)

but when trying to run in Python I still get the same error message:

Unrecognized tag family name: 'tagStandard32h11'. Families I know about:
tag36h11
tag25h9
tag16h5
tagCircle21h7
tagCircle49h12
tagStandard41h12
tagStandard52h13
tagCustom48h12

Is there something else i need to change/consider?

Horus26, 2021-06-18:
Solved the problem.

When installing I used the wrong conda environment, which did not have numpy installed.
When installing with the correct environment everything worked.

Thanks for your help. Maybe this issue can help other Python programmers aswell!

closed 2021-06-18 by Horus26