AprilRobotics/apriltag #153: how can i change rotation matrix to Euler Angle?

jcyhcs, 2021-06-18:
hi,professor:

i can easily get the rotation matrix use "pose.R->data[*]", then i want to change it to Euler Angle , so i can feed to UAV, but i don't know how to do,  and there is no reference, so ,please help me!

thanks a lot!

Reza666-cloud, 2021-06-18:
I meet same problem as you, and I solve it through above code:
`
         Mat rvec(3,3,CV_64FC1,pose.R -> data);
        Mat tvec(3,1,CV_64FC1,pose.t -> data);

        //calculate cam axis
        double r11 = rvec.ptr<double>(0)[0];
        double r21 = rvec.ptr<double>(1)[0];
        double r31 = rvec.ptr<double>(2)[0];
        double r32 = rvec.ptr<double>(2)[1];
        double r33 = rvec.ptr<double>(2)[2];

        double thetaz = atan2(r21,r11)/CV_PI *180;
        double thetay = atan2(-1*r31, sqrt(r32*r32+r33*r33)/CV_PI*180);
        double thetax = atan2(r32,r33)/CV_PI *180;

        double tx = pose.t ->data[0];
        double ty = pose.t ->data[1];
        double tz = pose.t ->data[2];
        double x = tx, y = ty, z = tz;

        codeRotateByZ(x, y, -1 * thetaz, x, y);
        codeRotateByY(x, z, -1 * thetay, x, z);
        codeRotateByX(y, z, -1 * thetax, y, z);
`

jcyhcs, 2021-06-21:
but i have some confused, the newest apriltag library, when i got rotation matrix, what the coordinate frame that the  matrix is relative to ?  
if there is a scene: the camera look down ground vertical, the camera matrix is right-hand frame, that Z axis forward direction is point to the ground.  just assume the apriltag's world coordinate frame , there Z axis forward direction is point to the sky, it also right-hand frame, then the Rotation Matrix describe how change the  origin point(in apriltag's world coordinate frame) to initial point( in camera coordinate frame)?

please help !

Reza666-cloud, 2021-06-21:
I think the rotation matrix and translation matrix from the newest apriltag lib are convert the world coordinate frame to camera coordinate frame. It means you are able to calculate the cam position in world coordinate frame through **three-dimensional coordinate system transformation**.

n-gineer, 2021-07-17:
> but i have some confused, the newest apriltag library, when i got rotation matrix, what the coordinate frame that the matrix is relative to ?
> if there is a scene: the camera look down ground vertical, the camera matrix is right-hand frame, that Z axis forward direction is point to the ground. just assume the apriltag's world coordinate frame , there Z axis forward direction is point to the sky, it also right-hand frame, **_then the Rotation Matrix describe how change the apriltag's world coordinate frame to camera coordinate frame?_**
> 
> please help !

This[ part of the user guide](https://github.com/AprilRobotics/apriltag/wiki/AprilTag-User-Guide#coordinate-system) should help.

closed 2021-08-09 by jcyhcs