AprilRobotics/apriltag #154: NaN in homography_compute2

TaoYibo1866, 2021-06-23:
https://github.com/AprilRobotics/apriltag/blob/b54fae744bdbddad62c4e0a0d39d8ba8c48a1d4c/apriltag.c#L476
During my stress test, process crashed and Visual Studio showed "stack around A crashed". I looked into the crash and found A[6] A[7] A[8], A[15] A[16] A[17], A[24] A[25] A[26], A[33] A[34] A[35], A[42] A[43] A[44], A[51] A[52] A[53], A[62], A[70], A[71] all equal -nan(id). Guess this is a bug?
[image]

TaoYibo1866, 2021-06-23:
The strange thing is this only occurs about 22min after the process started.

micology-code, 2022-02-22:
Same problem happened to me, some times it crash within 1 minute, some times it crash after half an hour

christian-rauch, 2022-02-22:
Is the input to this function (`double c[4][4]`) correct (finite values)? From the screenshot of the issue, it is hard to see what is going on and what all the values are.

Is there any chance this is reproducible? Can you share the input data and a script that reads the data and that eventually creates these conditions? Alternatively, can you share a gdb core dump for a known version of the library?

christian-rauch, 2022-05-19:
This is related to #72.

@TaoYibo1866 Does #213 solve your issue? Can you reproduce this with the example image linked in the PR?

TaoYibo1866, 2022-07-09:
Sorry, i am not able to reproduce this at present since the original project has been closed for a year. I am glad that there is PR to fix this. Thank you for your work.

closed 2022-07-09 by TaoYibo1866