AprilRobotics/apriltag #155: is there anybody can analyse all the codes, also with theory?

jcyhcs, 2021-06-24:
hi,professor:

your code is excellent! but my ability not enough, so when read this code, i always confused.
so can anybody explain this code  and theory behind one by one, then into documents?

that will be very helpful!

so , PLEASE!

mkrogius, 2021-06-24:
I would suggest reading the apriltag papers:

1. https://april.eecs.umich.edu/papers/details.php?name=olson2011tags
2. https://april.eecs.umich.edu/papers/details.php?name=wang2016iros
3. https://april.eecs.umich.edu/papers/details.php?name=krogius2019iros

The first paper is slightly out of date now, but has the most background information. The 2nd paper describes most of the ways that the first paper is slightly out of date. The third paper focuses on the newer tag families.

jcyhcs, 2021-06-25:
@mkrogius  hi,professor,
i study on the code, but i don't understand how can the code use four corner point in image , then caculate the four corner real 3D coordinate, i don't find in the 3 papers either, 
so , could you explain this question in detail? 
please help!

closed 2021-08-09 by jcyhcs