AprilRobotics/apriltag #156: Memory allocation

Lexicss2, 2021-06-24:
I would like just to ask regarding the source code.

In apriltag.c there is method called static void quick_decode_init(apriltag_family_t *family, int maxhamming)
and there is a logic for memory allocation. 
First, the parameter maxhamming == 2 by default. What chnages if I decrease it to 1 or even to 0.  I have a strict memory limitation on my side that is why I am looking the way hoe to decrees memory allocation. When I am doing it the memory allocation is significantly lower, and tag detection is working, but what the side affect of it?
Second, thete is code qd->nentries = capacity * 3; What is the magic number 3 and why we need this multiplition?

wxmerkt, 2021-06-24:
Related discussion: https://github.com/AprilRobotics/apriltag_ros/pull/93#issuecomment-815144047

mkrogius, 2021-06-24:
The maxhamming governs the number of cells in the tag that can be read incorrectly and still have the tag be detected correctly.

The multiplication by three is to create a hash table with a load factor of 33.3%. Setting the load factor higher may result in slower hash table lookups. However, we made the choice of 33.3% here as a choice that has to work for all tag families. If you know you are only going to be using one tag family, you could probably increase the load factor as long as you test that it isn't slowing things down to much.

closed 2022-03-10 by wxmerkt