AprilRobotics/apriltag #157: AprilTag edges refinement method

s-trinh, 2021-06-24:
Hi,

I am interested into the function that refines the corners location.

https://github.com/AprilRobotics/apriltag/blob/a0b37ad165d1908fc0c400898f28fb4bb16deb1b/apriltag.c#L740-L741

From these lines of code, "0" index is y-coordinate and "1" index is x-coordinate.

But here:

https://github.com/AprilRobotics/apriltag/blob/a0b37ad165d1908fc0c400898f28fb4bb16deb1b/apriltag.c#L764-L765

Looks like it is the opposite?

Is there any issue or is it just maybe a variable naming issue?

---

BTW, there is a paper comparing AprilTag 3, Apriltag C++, ArUco and OpenCV ArUco: [Determining and Improving the Localization Accuracy of AprilTag Detection](http://lewissoft.com/pdf/ICRA2020/0205.pdf).

It would also be great to have the possibility to detect partially occluded AprilTag marker, similarly to the Michael Kaess AprilTag.

QuentinTorg, 2021-12-20:
`(nx, ny)` is a vector perpendicular to the current edge.`(x0, y0)` is a point on the edge. They should be perpendicular to each other

If a vector pointing from quad corner a to quad corner b is `(x, y)`, then `(y, -x)` will be a vector perpendicular to it. In the first snippet, the `[0]` index is still the X coordinate, but the normal X coordinate is the original Y coordinate

s-trinh, 2021-12-28:
That makes sense, thanks!

closed 2021-12-28 by s-trinh