AprilRobotics/apriltag #161: Frequently no detection on clearly visible tags

n-gineer, 2021-07-17:
Clearly visible tags will sometimes be detected, and sometimes not at all. Shining a flashlight on the tag seemed to help, so I added a ring light to my raspberry pi, which did not seem to help. I've gotten detections on very bad lighting conditions, so I was not expecting such poor performance. 

Steps to reproduce the behavior:
Run apriltag detector on the attached input image. 

**Expected behavior**
Detects both tags

** Input Image**
[image]

**Operating Sytem**
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"

** Installation Method**
 I built AprilTag from source following the instructions in the README

** Code version **
 current github HEAD (v3.1.4).

**Additional context**
My intended usage is in the particular lighting scenario of an enclosed garage with one or two tags visible.

running with debug shows that the tag is not detected as a quad.

[debug_quads.pdf](https://github.com/AprilRobotics/apriltag/files/6835332/debug_quads.pdf)

The debug lines file seems to indicate the detector doesn't even recognize most of the lines on the tag.

[debug_lines.pdf](https://github.com/AprilRobotics/apriltag/files/6835333/debug_lines.pdf)

**EXAMPLE DETECTIONS IN POOR LIGHTING**

mkrogius, 2021-07-17:
Thanks for the interesting test case. I downloaded the image you have and got a slightly different result (after converting it to a .pnm file for use with the apriltag_demo binary). For me, the tag on the left was detected but the tag on the right was not.

The reason the tag on the right is not detected is because of the sharp lighting gradient on the bottom left corner. This causes the adaptive thresholding (see debug_threshold.pnm) to classify a chunk of the white border as black, which causes the derived edge (see debug_clusters.pnm) to loop all the way back around the outside of the tag.

[image]
[image]

There's two ways this could be fixed: a) by improving the adaptive thresholding or b) by detecting and deleting the parts of an edge that loop around likes this. I'm not sure what the best fix is, I'll have to think about it.

A more practical solution would be to remove any lamination from the tag or print it on something which is more matte and has less specular reflections.

n-gineer, 2021-07-18:
Thanks for the help debugging! I noticed that blur =2.0 went from being able to detect neither to both. I'm using the python bindings that came with the library, which I believe calls apriltag_demo. Apart from blur, I kept everything else defaulted.

Do you guys have a recommended printing method/product to make weatherproof tags? This tag is open to the weather pretty much any time other than my garage.

mkrogius, 2021-07-19:
The default settings may be different between the python interface and
apriltag_demo, which may explain the difference. it's interesting that blur
was able to fix the problem in the end.

In the past I have used bought tags printed on plastic from FedEx, but it
is more expensive than laminating paper.

christian-rauch, 2024-02-05:
Blurring the image helps.

With the converted jpeg image
[image]

and blur of 2 sigma:
```
./apriltag_demo -d -x 1 -b 2 126049372-14e3e0d3-6bb4-409c-915b-af4024f9ec3d.jpg
```
I get a detection for both tags:
```
detection   0: id (36x11)-576 , hamming 0, margin   46.279
detection   1: id (36x11)-576 , hamming 0, margin   38.775
```
[image]

closed 2024-02-05 by christian-rauch