AprilRobotics/apriltag #162: Questions concerning possible marker families

Horus26, 2021-07-27:
Hi again,

is it possible to create a tag family for example 12h11 with two possible ids?
Are there any disadvantages when using such a family as 12h11 compared to 32h11?
To my understanding, using 12h11 compared to 32h11 should result in a larger detection range, but less possible ids.

Thanks for your help.

mkrogius, 2021-07-27:
The main downside is that the false positive rate will be higher, because the ratio of the number of ids to the maximum possible ids (2^12) is much larger than the same ratio for the 32h11 family.

12h11 has a ratio of 2/2^12 ~= 5e-4
32h11 has a ratio of 500/2^32 ~= 1e-7
(i don't remember the exact number of ids) 

This means that the false positive rate will probably be about 500x larger.

Horus26, 2021-07-29:
Thanks for your answer!

closed 2021-07-29 by Horus26