AprilRobotics/apriltag #165: Unable to detect apriltag in image with illumination/reflection

mishti-29, 2021-08-12:
[https://stackoverflow.com/questions/68743775/remove-illumination-noise-from-image-to-detect-april-tag](url)
I am trying to detect apriltags in real-time live feed using apriltags library.
As per my knowledge,detection of apriltag is possible when we have clear edges .I believe that problem is due to the illumination / glare / reflection shown below in Figure 1 ,the edges are not clearly visible and I am unable to detect the tag .As I cannot implement heavy computation in my processor so firstly how to proceed with this problem or what are the possible solution for successful detection of tag?
and secondly , Does the "brightness due to light source" as shown in the Figure 2 leading to changes in saturation/contrast of image also causing problem in tag detection ? If yes, how does this brightness change  can affect tag detection  ?
I want to inform that I have tried to apply OTSU,Adaptive Thresholding to image but it did not help much in successful tag detection.
Figure 1:
[image]
Figure 2:
[image]

mkrogius, 2021-08-12:
I'd guess that the reason the tag in fig 1 is not detected is because of the weird artifact (lens flare?) on the right edge of the tag. Can you move/turn your camera so that the lens flare does not lie on the tag edge?

mishti-29, 2021-08-13:
No , the position of camera is fixed inside mechanism and glare can occur anywhere inside image in a live feed.So, Is there any function in apriltag library that can help detect apriltag in presence of this weird artifact (lens flare) or Is there any function/parameters that can help minimize the effect of lens flare while detecting apriltag?

mkrogius, 2021-08-13:
Detection should work as long as the artifact is not directly on the edge of the tag.

You could also try changing other parts of your imaging setup, like adding a lens hood/swapping lenses/changing filters.

closed 2021-12-17 by mkrogius