AprilRobotics/apriltag #166: binary bloom release for 3.1.4

christian-rauch, 2021-08-17:
Can we get a new bloom release for the current version 3.1.4 (or newer)?

I get people running into compatibility issues between the library and the ROS2 node because the namespace change in https://github.com/AprilRobotics/apriltag/commit/35f4ef2825745a08ec4690ce8c15f8dc903c2868 has not yet been released to the ROS repose via bloom.

wxmerkt, 2021-08-17:
I am not sure we should perform what looks like a patch release (3.1.2->3.1.4) but is likely to break downstream users without first restoring backwards compatibility. Would adding an alias for the current target without the namespace change such that both work be an option?

For context:

- v3.1.2 is currently released to foxy, galactic, rolling, melodic, and noetic.
- Changes between 3.1.2 and 3.1.4: https://github.com/AprilRobotics/apriltag/compare/3.1.2...v3.1.4
- The Python interface needs patches for the release (or to be turned off via a patch)

christian-rauch, 2021-08-17:
Well, while this is just a patch release by semver standards, it actually broke the compile-time dependency between 3.1.2 and 3.1.3 :-)

The packages are not backward or forward compatible, so I think the best is just to publish the library and node at the same time to the repo. Users of the binary packages will get both updated at the same time. Other users that compile from source will already use the newer "namespaced" version.

The issue is when users mix those, e.g. compile the node from source and use rosdep to resolve and install the binary library version. A new release will fix this. Do you have another scenario in mind where a new release will break something for others?

mkrogius, 2021-08-17:
I messed up here, by not really understanding what the cmake namespace change would do. My thought is that the best thing to do would be to have both the namespaced and non-namespaced versions and push a new version of apriltag (3.1.5). Would that fix the problems?

Also, mildly off-topic, it seems that I am becoming a bottleneck for addressing changes to this repository. I would be happy to give either of you access to this repository, if you would like, so that issues like this can be more quickly addressed.

wxmerkt, 2021-08-17:
> Do you have another scenario in mind where a new release will break something for others?

It might break things where people use the `apriltag` library but not the ROS nodes directly. Since AprilTag is widely used in the community, this is likely to be the case. It won't affect people who've been using PkgConfig, but if they've used the previous exported target they'd be affected.

> My thought is that the best thing to do would be to have both the namespaced and non-namespaced versions and push a new version of apriltag (3.1.5). Would that fix the problems?

My understanding is that it would - and I opened a PR with such a fix (#167). I am not sure if there might be unintended consequences of such an approach?

mkrogius, 2021-08-17:
This sounds good to me. I accepted the PR and bumped the version to 3.1.5 now.

christian-rauch, 2021-08-17:
> It might break things where people use the apriltag library but not the ROS nodes directly. Since AprilTag is widely used in the community, this is likely to be the case. It won't affect people who've been using PkgConfig, but if they've used the previous exported target they'd be affected.

If they use the upstream source releases, this would have broken their compile chain already. So I think this is only an issue for people that use the binary library in the ROS repo, but not the binary node - e.g. users of the ROS2 node that compile it from source.

> My understanding is that it would - and I opened a PR with such a fix (#167). I am not sure if there might be unintended consequences of such an approach?

I didn't know of this ALIAS flag before. If this allows using either the "namespaced" version or the "plain" version, then this makes sense to keep both for now. It would be useful to have some kind of warning in CMake, not sure if this is possible.

wxmerkt, 2021-08-17:
Thank you very much for the swift action @mkrogius :-).

I've prepared releases for
- **Melodic** (there was an issue with the patches, but it appears to have generated fine): https://github.com/ros/rosdistro/pull/30503
- **Noetic**: https://github.com/ros/rosdistro/pull/30501 
- **Foxy**: https://github.com/ros/rosdistro/pull/30502

There are some GitHub access issues when trying to prepare releases for Galactic/Rolling so will have to revisit this in the future

wxmerkt, 2021-08-19:
Galactic and Rolling are out now as well:
- **Galactic**: https://github.com/ros/rosdistro/pull/30518
- **Rolling**: https://github.com/ros/rosdistro/pull/30519

Unfortunately the fix in #167 does not work with the CMake version shipped on Ubuntu 18.04/ROS Melodic. It does work fine on 20.04. I'll investigate and make a follow-up fix/PR/release.

wxmerkt, 2021-08-19:
A release for 3.1.6 containing the hotfix for old CMake versions is now out for Melodic (https://github.com/ros/rosdistro/pull/30520) - I'll close this issue, please ping if there are remaining issues

closed 2021-08-19 by wxmerkt

Tobias-Fischer, 2021-09-16:
Hi @wxmerkt - any chance to release 3.1.6 for noetic?

wxmerkt, 2021-09-17:
@Tobias-Fischer apologies, this was an oversight not to release for Noetic. It's done now: https://github.com/ros/rosdistro/pull/30812

Tobias-Fischer, 2021-09-18:
Thanks @wxmerkt!