AprilRobotics/apriltag #168: Static measurement error with some lenses from apriltag_ros

halil54, 2021-08-18:
Hi there,

I am using apriltags with different cameras and lenses. For some lenses there is no problem with the measurements ı get really accurate results. However, for some lenses there is error and its ratio is same for all distances, for example yesterday I tried a new lens, results were 0.8 of real distances. I am using ros camera calibration, same procedure for all lenses. I would doubt the calibration procedure I am following If I didn't get really accurate results for some lenses.
Any help would be really appreciated.
Thanks in advance.

christian-rauch, 2021-08-18:
To make sure, are you calibrating each camera-and-lens configuration separately?

I usually get good results after calibration. But I am using cameras with relative narrow field-of-views (<=90°) and low distortion. What kind of lenses are you using? Do the lenses that give you a bad calibration have higher distortion or field-of-views?

I am not sure if the `cameracalibrator.py` calibration script supports all of the OpenCV parameters of the calibration function [`cv::calibrateCamera`](https://docs.opencv.org/4.5.2/d9/d0c/group__calib3d.html#ga3207604e4b1a1758aa66acb6ed5aa65d). You see there for example that some of the distortion parameters (k4, k5, k6 and s1, s2, s3, s4) are only used when the flags `CALIB_RATIONAL_MODEL` and `CALIB_THIN_PRISM_MODEL` are set. I would recommend exploring the parameters of `cameracalibrator.py` and if it does not mirror all the required function parameters, then I would try to do this manually with the underlying OpenCV functions.

halil54, 2021-08-18:
Hi,
Yes I am making calibrations for each set separately. I am also using narrow field of view and low distortion lenses. Today I have tried one 80 hfov runcam lens, and one lens from arducam similar fov. Both of them gave me same approximately same results. There is a static ratio btw real and apriltag distances which is around 1.2, I am considering to change the tag size from the config so that the results will be accurate. Do you think this would cause some problems for something else ? 

As I said before there are some other lenses with similar field of views giving me really accurate results. So there shouldn't be any problem with the calibration procedure. I need these lenses for apriltag detection with IR illumination.

mkrogius, 2021-08-18:
Can you double check you have measured the tag size as described in the user guide?

https://github.com/AprilRobotics/apriltag/wiki/AprilTag-User-Guide#pose-estimation

halil54, 2021-08-18:
I am using 48h12custom tag, I have double checked right now tag size is correct.

christian-rauch, 2021-08-18:
Having a constant "error" sounds indeed like using the wrong size of the calibration pattern or tag. If you want to be sure that the calibration is correct, you could try to measure the distance and size of an object in the real world and compare this to the distances in the image/sensor and the focal length (both in pixel or metric size). The ratios should be very similar.

halil54, 2021-08-19:
It seems that I have found the problem. Lenses which give me measurement error have high distortion coefficients, though I didn't see much distortions in the images and their hfov is around 80 degree. What can be the cause of this high distortion coefficient ? 

Camera calibration results of one of the problematic lens:
image_width: 1920
image_height: 1080
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [ 1431.13739,     0.     ,   947.32253,
             0.     ,  1440.1271 ,   536.14141,
             0.     ,     0.     ,     1.     ]
camera_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.388768, 0.165661, 0.001307, -0.004290, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [ 1.,  0.,  0.,
          0.,  1.,  0.,
          0.,  0.,  1.]
projection_matrix:
  rows: 3
  cols: 4
  data: [ 1160.61609,     0.     ,   925.4137 ,     0.     ,
             0.     ,  1357.33557,   536.65778,     0.     ,
             0.     ,     0.     ,     1.     ,     0.     ]
             
             
From the following one I am getting really accurate results.

image_width: 1280
image_height: 720
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [ 942.21775,    0.     ,  653.27891,
            0.     ,  938.44487,  346.27151,
            0.     ,    0.     ,    1.     ]
camera_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.031221, -0.064961, 0.003057, -0.002078, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [ 1.,  0.,  0.,
          0.,  1.,  0.,
          0.,  0.,  1.]
projection_matrix:
  rows: 3
  cols: 4
  data: [ 943.19464,    0.     ,  650.85265,    0.     ,
            0.     ,  943.49542,  347.86372,    0.     ,
            0.     ,    0.     ,    1.     ,    0.     ]

closed 2022-01-27 by christian-rauch