AprilRobotics/apriltag #171: Memory leak

afdaniele, 2021-09-10:
**Describe the bug**

Memory allocation seems to grow unbounded.

**To Reproduce**

Build `apriltag 3.1.6` and run the following code snippet on the image attached (I tried with other images as well and the result is the same).

```
import time
from apriltag import apriltag
from PIL import Image, ImageOps

# load image
image = Image.open("./image.png")
gray = ImageOps.grayscale(image)

# detect tags (forever)
while True:
    detector = apriltag("tag36h11")
    detections = detector.detect(gray)
    time.sleep(1.0 / 120.0)
```

**Expected behavior**

Memory allocation of the process converges to a stable value.

**Input Image**

[image]

**Operating Sytem**

Tested on two systems:

- Ubuntu 20.04 (amd64)
- Ubuntu 20.04 (arm64v8) - Raspberry Pi 4 4GB

**Installation Method**

I built AprilTag (and the Python bindings) from source following the instructions in the README.
A different version of Python bindings for `apriltag` that I'm maintaining (https://github.com/duckietown/lib-dt-apriltags)
has the same issue. 

**Code version**

Commit: https://github.com/AprilRobotics/apriltag/commit/c515a79b1364e2d48a0c839d8ed170646a837524
Tag: v3.1.6

Also tested tag v3.1.1 and have the same issue.

**Additional context**

The problem seems to be in the C code of the main library and not in the Python bindings.
This is due to the fact that completely different Python bindings (https://github.com/duckietown/lib-dt-apriltags) are affected by the same problem.

Running the code above, which runs detection at a max speed of `120Hz`, produces a process that starts at about `100MB` of memory allocated and grows consistently over time with a rate of about `245MB / hour`.

mkrogius, 2021-09-16:
Please confirm that this change fixed the issue for you.

closed 2021-09-16 by mkrogius

afdaniele, 2021-09-17:
@mkrogius 
I kept this running for 24 hours and the memory usage converged to ~230MB after a while, but never exploded.
So, I guess we can consider this solved.
Thanks.

mkrogius, 2021-09-18:
Glad to hear it, and thanks for checking on the fix. I'm glad to have gotten this one fixed, since it was in a part of the code that I haven't looked at much.

commit 80bf7320 (2021-09-16): Fix ref-counting issue in python wrapper. Fixes #171