AprilRobotics/apriltag #174: Can not use the Tag Family in Python 3 after adding the .c and .h files and reinstalling

yusufguleray, 2021-09-24:
I would like to use the tag family `'tag36h10'` in my Python 3 program but even after adding the `'tag36h10.c'` and `'tag36h10.h'` and reinstalling with
```
$ cmake .
$ sudo make install
```
I get the
 
```
import apriltag
detector = apriltag.apriltag('tag36h10')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Unrecognized tag family name: 'tag36h10'. Families I know about:
  tag36h11
  tag25h9
  tag16h5
  tagCircle21h7
  tagCircle49h12
  tagStandard41h12
  tagStandard52h13
  tagCustom48h12
```
How can fix this so that I can use 'tag36h10' in my Python 3 program?

Note: Here is the output of my installation
```
$ cmake .
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yus/MasterThesis/libraries/apriltag

$ sudo make install
Scanning dependencies of target apriltag
[  2%] Building C object CMakeFiles/apriltag.dir/apriltag.c.o
[  5%] Building C object CMakeFiles/apriltag.dir/apriltag_pose.c.o
[  7%] Building C object CMakeFiles/apriltag.dir/apriltag_quad_thresh.c.o
[ 10%] Building C object CMakeFiles/apriltag.dir/common/g2d.c.o
[ 13%] Building C object CMakeFiles/apriltag.dir/common/getopt.c.o
[ 15%] Building C object CMakeFiles/apriltag.dir/common/homography.c.o
[ 18%] Building C object CMakeFiles/apriltag.dir/common/image_u8.c.o
[ 21%] Building C object CMakeFiles/apriltag.dir/common/image_u8x3.c.o
[ 23%] Building C object CMakeFiles/apriltag.dir/common/image_u8x4.c.o
[ 26%] Building C object CMakeFiles/apriltag.dir/common/matd.c.o
[ 28%] Building C object CMakeFiles/apriltag.dir/common/pam.c.o
[ 31%] Building C object CMakeFiles/apriltag.dir/common/pjpeg-idct.c.o
[ 34%] Building C object CMakeFiles/apriltag.dir/common/pjpeg.c.o
[ 36%] Building C object CMakeFiles/apriltag.dir/common/pnm.c.o
[ 39%] Building C object CMakeFiles/apriltag.dir/common/string_util.c.o
[ 42%] Building C object CMakeFiles/apriltag.dir/common/svd22.c.o
[ 44%] Building C object CMakeFiles/apriltag.dir/common/time_util.c.o
[ 47%] Building C object CMakeFiles/apriltag.dir/common/unionfind.c.o
[ 50%] Building C object CMakeFiles/apriltag.dir/common/workerpool.c.o
[ 52%] Building C object CMakeFiles/apriltag.dir/common/zarray.c.o
[ 55%] Building C object CMakeFiles/apriltag.dir/common/zhash.c.o
[ 57%] Building C object CMakeFiles/apriltag.dir/common/zmaxheap.c.o
[ 60%] Building C object CMakeFiles/apriltag.dir/tag16h5.c.o
[ 63%] Building C object CMakeFiles/apriltag.dir/tag25h9.c.o
[ 65%] Building C object CMakeFiles/apriltag.dir/tag36h10.c.o
[ 68%] Building C object CMakeFiles/apriltag.dir/tag36h11.c.o
[ 71%] Building C object CMakeFiles/apriltag.dir/tagCircle21h7.c.o
[ 73%] Building C object CMakeFiles/apriltag.dir/tagCircle49h12.c.o
[ 76%] Building C object CMakeFiles/apriltag.dir/tagCustom48h12.c.o
[ 78%] Building C object CMakeFiles/apriltag.dir/tagStandard41h12.c.o
[ 81%] Building C object CMakeFiles/apriltag.dir/tagStandard52h13.c.o
[ 84%] Linking C shared library libapriltag.so
[ 84%] Built target apriltag
Scanning dependencies of target apriltag_python
[ 86%] Generating apriltag.cpython-38-x86_64-linux-gnu.so
[ 94%] Built target apriltag_python
Scanning dependencies of target apriltag_demo
[ 97%] Building C object CMakeFiles/apriltag_demo.dir/example/apriltag_demo.c.o
[100%] Linking C executable apriltag_demo
[100%] Built target apriltag_demo
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/include/apriltag/apriltag.h
-- Up-to-date: /usr/local/include/apriltag/apriltag_math.h
-- Up-to-date: /usr/local/include/apriltag/apriltag_pose.h
-- Up-to-date: /usr/local/include/apriltag/tag16h5.h
-- Up-to-date: /usr/local/include/apriltag/tag25h9.h
-- Up-to-date: /usr/local/include/apriltag/tag36h10.h
-- Up-to-date: /usr/local/include/apriltag/tag36h11.h
-- Up-to-date: /usr/local/include/apriltag/tagCircle21h7.h
-- Up-to-date: /usr/local/include/apriltag/tagCircle49h12.h
-- Up-to-date: /usr/local/include/apriltag/tagCustom48h12.h
-- Up-to-date: /usr/local/include/apriltag/tagStandard41h12.h
-- Up-to-date: /usr/local/include/apriltag/tagStandard52h13.h
-- Up-to-date: /usr/local/include/apriltag/common/doubles.h
-- Up-to-date: /usr/local/include/apriltag/common/doubles_floats_impl.h
-- Up-to-date: /usr/local/include/apriltag/common/floats.h
-- Up-to-date: /usr/local/include/apriltag/common/g2d.h
-- Up-to-date: /usr/local/include/apriltag/common/getopt.h
-- Up-to-date: /usr/local/include/apriltag/common/homography.h
-- Up-to-date: /usr/local/include/apriltag/common/image_types.h
-- Up-to-date: /usr/local/include/apriltag/common/image_u8.h
-- Up-to-date: /usr/local/include/apriltag/common/image_u8x3.h
-- Up-to-date: /usr/local/include/apriltag/common/image_u8x4.h
-- Up-to-date: /usr/local/include/apriltag/common/matd.h
-- Up-to-date: /usr/local/include/apriltag/common/math_util.h
-- Up-to-date: /usr/local/include/apriltag/common/pam.h
-- Up-to-date: /usr/local/include/apriltag/common/pjpeg.h
-- Up-to-date: /usr/local/include/apriltag/common/pnm.h
-- Up-to-date: /usr/local/include/apriltag/common/postscript_utils.h
-- Up-to-date: /usr/local/include/apriltag/common/string_util.h
-- Up-to-date: /usr/local/include/apriltag/common/svd22.h
-- Up-to-date: /usr/local/include/apriltag/common/time_util.h
-- Up-to-date: /usr/local/include/apriltag/common/timeprofile.h
-- Up-to-date: /usr/local/include/apriltag/common/unionfind.h
-- Up-to-date: /usr/local/include/apriltag/common/workerpool.h
-- Up-to-date: /usr/local/include/apriltag/common/zarray.h
-- Up-to-date: /usr/local/include/apriltag/common/zhash.h
-- Up-to-date: /usr/local/include/apriltag/common/zmaxheap.h
-- Installing: /usr/local/lib/libapriltag.so.3.1.0
-- Up-to-date: /usr/local/lib/libapriltag.so.3
-- Up-to-date: /usr/local/lib/libapriltag.so
-- Installing: /usr/local/share/apriltag/cmake/apriltagTargets.cmake
-- Installing: /usr/local/share/apriltag/cmake/apriltagTargets-release.cmake
-- Installing: /usr/local/share/apriltag/cmake/apriltagConfig.cmake
-- Installing: /usr/local/share/apriltag/cmake/apriltagConfigVersion.cmake
-- Installing: /usr/local/lib/pkgconfig/apriltag.pc
-- Installing: /home/yus/.local/lib/python3.8/site-packages/apriltag.cpython-38-x86_64-linux-gnu.so
-- Installing: /usr/local/bin/apriltag_demo
-- Set runtime path of "/usr/local/bin/apriltag_demo" to ""
```

mkrogius, 2021-09-24:
Try adding the family name to the list here: https://github.com/AprilRobotics/apriltag/blob/master/apriltag_pywrap.c#L20

yusufguleray, 2021-09-25:
Thank you for your answer, @mkrogius !
So as you suggested :

1. adding the tag family name to [here](https://github.com/AprilRobotics/apriltag/blob/master/apriltag_pywrap.c#L20) (in this case : `    _(tag36h10)                             \` ) 
2. and also including .h file of the tag family to [here](https://github.com/AprilRobotics/apriltag/blob/9fd619b528b2792bac89576b1e1713c6807eeaf7/apriltag_pywrap.c#L18), (in this case : `#include "tag36h10.h"`)

fixed the issue for me.

closed 2021-09-25 by yusufguleray

christian-rauch, 2021-09-25:
@yusufguleray Can you send a PR with the change that fixes your issue?