AprilRobotics/apriltag #176: is this paper's method can improve detect apriltags more accurate?

jcyhcs, 2021-09-28:
hi,professor:
i find a paper 《Determining and Improving the Localization Accuracy of AprilTag Detection》2020，ICRA。
it mention an improvement for apriltags, extract some words in the paper:
[image]

can this method working? if it worth trying, can realize this improvement in this git project?
thanks for your help!

mkrogius, 2021-09-30:
It may be an improvement, but I'm not sure. The problem is that all the data in the paper is from simulation and I'm not sure that the results would transfer into the real world.

I also don't have time right now to implement it.

s-trinh, 2021-10-03:
I have tried to reproduce the "Determining and Improving the Localization Accuracy of AprilTag Detection" paper results but without success. Please find [here](https://github.com/UniBwTAS/apriltags_tas/issues/4) the code I am using to benchmark and the data.

Maybe there is an obvious error you could easily spot.

---

As a side note, the AprilTag 3 corner refinement method works pretty well and is easy to integrate since it does not require extra library. Thanks for distributing the AprilTag code with a permissive license.