AprilRobotics/apriltag #180: Not stable for tag16h5?

libing64, 2021-11-09:
**Describe the bug**
The detection result is not stable, is that a bug? 

[image]

**To Reproduce**
Steps to reproduce the behavior:
```
./opencv_demo -f tag16h5
```

```
        imshow("Tag Detections", frame);
        char key = waitKey(30);
        if (key == 'q')
            break;
```

**Expected behavior**
A clear and concise description of what you expected to happen.

** Input Image**
If applicable, please attach the input image that reproduces the problem.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating Sytem**
[e.g. Ubuntu, iOS, Windows]

** Installation Method**
[e.g. I built AprilTag from source following the instructions in the README]

** Code version **
What does the version tag in package.xml say? Or, if you have downloaded the code from github right now, say current github HEAD.

**Additional context**
Add any other context about the problem here.

mkrogius, 2021-11-09:
It is most likely not a bug. The tag16h5 family has an extremely high false positive rate compared to the other tag families

libing64, 2021-11-10:
@mkrogius Are there any suggestions to optimize this problem?

mkrogius, 2021-11-10:
I suggest using a different tag family. Either tagStandard41h12 or tag36h11

libing64, 2021-11-11:
> I suggest using a different tag family. Either tagStandard41h12 or tag36h11

Got it, great thanks.

closed 2021-11-11 by libing64