AprilRobotics/apriltag #181: Invalid rotation matrices from estimate_tag_pose()

DavidRisch, 2021-11-12:
**Describe the bug**
For some rare cases `estimate_tag_pose_orthogonal_iteration()` and by extension `estimate_tag_pose()` produce invalid rotation matrices with a determinant of -1.
This bug does not happen when only using `estimate_pose_for_tag_homography()`, so it seems to be located in `orthogonal_iteration()`. It does not happen when running `estimate_tag_pose_orthogonal_iteration()` with `nIters` = 0.
An error message is printed to stderr: `Error, more than one new minima found.`

**To Reproduce**
Steps to reproduce the behavior:
Run this modified version of opencv_demo.cc on the attached input image: https://gist.github.com/DavidRisch/b7d2b39c2d240954ab63207a62879b51
Modifications made to opencv_demo.cc:
- An image is read from a path given in the arguments instead of a video input
- The pose of each detected tag is estimated
- Print the estimated rotation matrix and its determinant
- The estimated pose is drawn in the image

**Expected behavior**
I expect `estimate_tag_pose_orthogonal_iteration()` and by extension `estimate_tag_pose()` to return a valid rotation matrix (R^T = R^-1 and det(R) = 1).

**Input Image**
[image]

**Screenshots**
Error: With `estimate_tag_pose_orthogonal_iteration(..., 1)` (all other `nIters` values > 0 seem to prdoduce diffrent but also incorrect results):
[image]

Expected behavior: With `estimate_tag_pose_orthogonal_iteration(..., 0)` or `estimate_pose_for_tag_homography(..)` (both produce the same image):
[image]

**Operating Sytem**
openSUSE Leap 15.1

**Installation Method**
I built AprilTag from source following the instructions in the README

**Code version**
Current github HEAD: b1b80fe37d4454b3333dc69b6edb353092e4a8c1
I originally used version 3.1.1 and experienced the same problem.

**Additional context**
none

mkrogius, 2021-11-13:
Thank you for providing the code + images, it made it very easy for me to test out this problem.

When I test this problem on current master, I get the result below which looks correct to me (I also tried with nIters=100 with the same result). This makes sense because I thought this issue had already been fixed in https://github.com/AprilRobotics/apriltag/issues/104

[image]

DavidRisch, 2021-11-15:
You are totally right. I apparently messed up linking my script against the new apriltag library and thus only ever tested with the old version. After fixing this I get the same result as you with the new version.
Thanks for the quick help!

closed 2021-11-15 by DavidRisch