AprilRobotics/apriltag #182: Calls to exit() in apriltag.c

j-medland, 2021-11-28:
Hello

I have been developing a wrapper to allow for AprilTag to be integrated into other applications as a DLL/shared library.

During this development I have encountered a call to `exit()` which will terminate the whole application; an example being in the internal call `quick_decode_init` when adding a tag family to a detector.

I could overcome this in my code but would you be interested in changing this behaviour in your library?

I would be happy to contribute a PR but I don't know if you are interested or have an alternative approach in mind.

Cheers

John

mkrogius, 2021-11-28:
Hi,

I think that would be a good improvement to the library. Instead of `exit`ing, `quick_decode_init` and `apriltag_add_family_bits` could both return an error code to indicate success or failure. If you contribute a PR, I would review it.

Best,
Max

christian-rauch, 2021-11-29:
I also think this would be a nice contribution. A library should in general not control the exiting of the program.

j-medland, 2021-12-01:
Following discussion on #183 

I am happy to take a look at removing the `exit()` and `printf()` calls in the library.

With regards to `exit()`, I can find three calls:

1. `apriltag_detector_add_family*` calls can be modified from  a `void` to integer `return` which indicates success/failure when the internal `quick_decode_init` succeeds or fails.
2.  `workerpool_create` used by `apriltag_detector_detect` currently handles failure by exiting. The only way I could see to indicate that `apriltag_detector_detect` failed due to `workerpool_create` in a non-breaking way would be to return an empty detections `zarray` and set `errno` to something like `EAGAIN`
3.  `g2d_polygon_contains_point`  called by `apriltag_detector_detect` has the dead code branch which could just be removed
```c
    if (0 && v != g2d_polygon_contains_point_ref(poly, q)) {
        printf("FAILURE %d %d\n", v, quad_acc);
        exit(-1);
    }
```
With regards to `printf()` - most are easy to fix but;

1. Would you like to comment out the `printf` lines or should they be deleted?
2. `quick_decode_init` prints an error against hamming values over 3 but doesn't actually exit. I suggest we could return a failure from the function call and set `errno` to `EINVAL` or maybe `ERANGE`.

Let me know what you think

christian-rauch, 2022-03-11:
@j-medland Are you still interested in removing the `exit` calls in the library code? I am still interested in this and will be happy to review a PR that replaced those with error codes or similar.

christian-rauch, 2022-05-19:
Fixed by https://github.com/AprilRobotics/apriltag/pull/224.

closed 2022-05-19 by christian-rauch