AprilRobotics/apriltag #184: segfault in multithreaded use

fdarvas, 2021-12-03:
I am seeing some frequent random segfaults coming from here: 

https://github.com/AprilRobotics/apriltag/blob/76482356ff7ecb3f4895b3b600d533b2e65980d9/apriltag.c#L188
when running it with 48 threads. 

Switching to a single thread seems to work fine. 

I am using the pupil Apriltags python wrapper, but that just fails in apriltag_detector_destroy, which in turn causes the segfault in         quick_decode_uninit(fam);

repeatedly run the detector with a large number of threads (e.g. 48)

**Expected behavior**
I would expect this to just cleanly exit.

**Operating Sytem**
Ubuntu 20.04 WSL

** Installation Method**
pip install of pupil_apriltags
** Code version **
3.1.0
**Additional context**
Add any other context about the problem here.

christian-rauch, 2021-12-03:
Can you share a full gdb backtrace? If possible, can you also share the output of valgrind's thread error detector (`valgrind --tool=helgrind`)?

I see you ran this on Windows via WSL. Can you try to run this on bare metal?

fdarvas, 2021-12-04:
I think I got rid of it by moving the detector out of the python object - initially I had an object, that would set up the detector once and load a bunch of images, then run the detector in one of the methods - upon deletion of the object it would then crash - now that I moved it to a separate function outside of the object, it seems to be doing fine. 

Here is the gdb trace for the segfault:
[gdb.txt](https://github.com/AprilRobotics/apriltag/files/7653023/gdb.txt)

christian-rauch, 2021-12-06:
> Here is the gdb trace for the segfault:

There is no backtrace in the log. Can you wait for the crash and the type `bt` + Enter to show the backtrace? Also you have to compile the library with debug symbols. When you provide the backtrace, also specify on which commit this was generated. So that we can associate the source references in the backtrace with the code.

christian-rauch, 2022-06-28:
Closing this as this issue seems to have disappeared and we do not have any information to debug it. The implementation for threading also change a bit, so this issue may not be present anymore.