AprilRobotics/apriltag #190: No quad detection after seemingly good cluster detection

huberya, 2021-12-16:
Hi
I've been using Apriltag2 for a while now, with a detector from ASL and wanted to switch to Apriltag3 and using the built in detector.

Note that I use a two pixel border for the 36h11 family and thus applied the "hack" proposed in #45.

I turned on "debug" and set  "quad_decimate" to 1.0 and also varied a bit with the "blur" without success.

My initial image is:
[image]

The clusters image still looks good from what I understand:
[image]

But then the in the next output, if I followed the code correctly it seems to discard all quads (debug_quads_raw):
[image]

Any ideas on why this happens or what I could do to achieve better detections.

Thanks already for your help

mkrogius, 2021-12-17:
This setup was supported in AprilTag 1, but not Apriltag 2 or 3. See https://github.com/AprilRobotics/apriltag/issues/141 for details/discussion

huberya, 2021-12-17:
Ah I did not see that, thanks

closed 2021-12-17 by huberya