AprilRobotics/apriltag #191: Some questions about the accuracy of AprilTag

Bowangjian, 2021-12-22:
Hello, I am a graduate student. I have the following questions in the process of using AprilTag. I hope to get your help:
1. When multiple tags are used for pose estimation, should the weighted average of each independent pose be used or the pose obtained by SolvePnPRansac for the detected label corners is more accurate?
2. When there is a certain distance between the label and the camera, is the number of labels in the field of view the better, or the larger the label, the better? How to make a balance? Do you have any good suggestions?
3. Does the label's position in the camera's field of view affect the accuracy of pose estimation? For example, the label is in the center or corner of the image
4. I have seen some papers that state that the best pose estimation result will be obtained when the optical axis of the camera points to the center of the label during movement, but there are also some papers that write that when it is facing the center of the label, There will not be an optimal pose estimation result. On the contrary, a slight deviation from the center of the label will result in a higher pose result. What do you think?
Looking forward to your views and answers

christian-rauch, 2021-12-22:
> When multiple tags are used for pose estimation

Do you mean multiple pose estimations on individual tags in parallel or a single pose estimation on a tag bundle? It is possible to define a group that is made of multiple tags and estimate a single pose for this group.

> When there is a certain distance between the label and the camera, is the number of labels in the field of view the better, or the larger the label, the better?

Here it is unclear if you want to estimate a single pose from a tag bundle or multiple poses from individual tags. For a single pose estimation on a single tag, it's of course better to make it as large as possible. The advantage of a tag bundle with multiple tags is that a subset of tags can be occluded and the pose estimation will still be possible. This is not possible when doing this on a single tag and this single tag is partially occluded.

> Does the label's position in the camera's field of view affect the accuracy of pose estimation?

In theory not. But in practice, you will have some image distortion on the boundaries/edges of images, even after calibration. This is especially prominent with cameras that have a large field-of-view. So unless you have a perfect calibration, try to keep the tag close to the centre This way, the edge detection etc. will be less affected by distortions.

> I have seen some papers that state ...

I will be surprised if the pose estimation results from the image boundaries will be worse than from the image centre.

Bowangjian, 2021-12-22:
> Do you mean multiple pose estimations on individual tags in parallel or a single pose estimation on a tag bundle? It is possible to define a group that is made of multiple tags and estimate a single [...]

Thank you for your prompt reply. My biggest doubt lies in:
1. At a long distance, for example, the distance between the camera and the tag is 2m. Is the size of a single tag of 50cm better than a tag beam composed of 3cm and composed of 10 tags, this How to make a choice? The key is how to make a choice? What should be the basis?
2. On the third question, your reply is that if you use a camera with a large viewing angle, you should try to ensure that the detected label is in the center of the image to avoid being affected by distortion. But for the fourth question, what does your answer mean? It is because the label is close to the image boundary that the pose estimation effect will be better. I am a little confused, because there are two seemingly contradictory statements. Can you explain further?
3. The first question above is that when using the tag bundle for positioning, his pose solution strategy is to drop the four corner points of all the tags detected in the solvepnp() function parameter. This is confirmed The optimal solution strategy? Is there any relevant material description?
Once again, thank you very much for your answer, your answer has benefited me a lot, and I look forward to your new reply!

christian-rauch, 2021-12-22:
> Is the size of a single tag of 50cm better than a tag beam composed of 3cm and composed of 10 tags

I haven't done this comparison for myself. I cannot give you a good answer to this. *I think* using one large tag is better than using multiple small tags because the pose estimation on a larger tag is more reliable. But I will be happy if someone can show me otherwise and maybe support this with some data. Sometimes your setup just does not allow sticking a large tag on the object you want to track and you have to use multiple smaller tags in a bundle.

> On the third question, your reply is that if you use a camera with a large viewing angle, you should try to ensure that the detected label is in the center of the image to avoid being affected by distortion. But for the fourth question, what does your answer mean?

My argument is that the pose estimation in the image centre will be better than on the outer edges because the outer boundaries will have some distortion. The larger the FoV, the larger the distortion will be. You can reduce this distortion by calibrating and rectifying the image. But unless your calibration and rectification are perfect, you will still have some small (maybe neglectable) distortion.

In your question about contradicting statements from papers, I meant that I also think that pose estimation in the centre will be more accurate. I originally wrote
>> I will be surprised if the pose estimation results from the image boundaries will be worse than from the image centre.

but this is of course contradicting. I meant this the other way around. I fixed my original comment.

> The first question above is that when using the tag bundle for positioning, his pose solution strategy is to drop the four corner points of all the tags detected in the solvepnp() function parameter. This is confirmed The optimal solution strategy?

I don't understand this question. Is it still which approach (small tag bundles vs. large single tag) is better or how the tag bundle detection works? I've only used tag bundles when I expected that one of the tags from the bundle will be occluded. I don't know if this is better than using a single large tag or how this is implemented. Sorry.

Bowangjian, 2021-12-22:
> I haven't done this comparison for myself. I cannot give you a good answer to this. _I think_ using one large tag is better than using multiple small tags because the pose estimation on a larger tag [...]

Thank you very much for your prompt reply, which answered a lot of my doubts. One of my current work for my graduation project is to study the trade-off between the number of labels and the size of labels when choosing labels. Because my application scene is a scene with large distance changes, how to choose a suitable label layout at different distances? What are your views or suggestions on this?
My last question is more precise: if 4 sub-labels in a label bundle are detected, then the pose calculated by the label bundle (using 4 * 4 corner points) and the four sub-labels (using only 4 corner points) the pose calculated by each, and then process the results of these four poses, such as weighted average, will it be more accurate? The point of doubt is how to deal with multiple labels in the field of vision. What is your opinion?
Another question is: Regarding the influence of the relative pose (RPY) relationship between the camera and the tag on the pose estimation, I am really puzzled. For example, the viewpoint mentioned in the paper "Accuracy analysis of marker-based 3D visual localization": When the camera is facing the tag, there will not be the best pose estimation effect. On the contrary, there is a slight posture offset between the camera and the tag, which will have a more effective pose estimation effect. I have written this argument in at least three articles. The article mentioned, a general expression: This phenomenon is interesting and counter-cognitive. What do you think?
Looking forward to your reply! Your reply has benefited me a lot!

christian-rauch, 2021-12-22:
That's a lot of questions. I already stated my view on using a single large tag vs. multiple small tags in a tag bundle.

I don't have an answer to your question about which approach (single 4 corners vs. 4 x 4 corners in tag bundles) gives you a more accurate pose estimation. Since this is your graduation project, I suggest you create a setup with these two settings and do the evaluation compared to some ground truth. If you have a (calibrated) RGB-D camera, you can also use the depth to get some idea about the error.

I haven't read the article you mentioned. My experience is that tags closer to the centre give you a better pose estimation result.

Since those questions are not about the library itself, but about the general problem of tag detection, I would close this. I suggest asking such questions on https://robotics.stackexchange.com. You will get more opinions from others and the discussion will also be better visible. There are also other libraries for AprilTag detection and pose estimation. It is worth also comparing to these.

closed 2021-12-22 by christian-rauch

LordonCN, 2024-08-30:
> 4\. but there are also some papers that write that when it is facing the center of the label, There will not be an optimal pose estimation result. On the contrary, a slight deviation from the center of the label will result in a higher pose result.

It's true that when it is facing the center of the label the error of heading can be 10°+, but it will reduce to 1° when a slight deviation from the center of the label. May be due to the gimbal lock.