AprilRobotics/apriltag #201: Apriltag detection evaluation reports

hakanErgin, 2022-01-31:
Hi, we want to use apriltags in our project for pose estimation in possibly long distances and on moving target. 
I am looking for some evaluation reports that would provide some insight on questions like;
-at what distances can we detect tags, and what should be the resolution of these images?
-at what speeds will the tag be still detected?
And other similar potential questions that can apply to a moving distant (at max distance) target.
I thought I'd find some papers evaluating benchmarks, but I could not manage to find much else than comparisons between different tag libraries (which are not focusing on what we're looking for), benchmarks of close range applications, different light conditions, etc.

I'd be very happy to know if there are papers making this analysis. 
Thanks

christian-rauch, 2022-01-31:
See https://github.com/AprilRobotics/apriltag/issues/189 and https://github.com/AprilRobotics/apriltag/issues/191 for related discussions.

For general questions about AprilTags that are unrelated to the specific implementation in this library, I recommend https://robotics.stackexchange.com.

mkrogius, 2022-01-31:
+1, I'll also add that you can try out apriltags using the iphone app if
you want a quick way to test it

hakanErgin, 2022-01-31:
Thank you both. I have actually tried the ROS implementation already, but I was looking for some tests results which I was sure that exist : )
[Here](https://robotics.stackexchange.com/questions/23177/looking-for-benchmarks-on-apriltag-pose-estimation) I have opened a thread.

closed 2022-01-31 by hakanErgin