AprilRobotics/apriltag #222: Hi,some questions

Zhang-Qinghui, 2022-03-17:


Zhang-Qinghui, 2022-03-17:
Hi,Thanks for your great work! I have some questions when I read your article "AprilTag: A robust and flexible visual fiducial system". In Setion " Homography and extrinsics estimation", why the  camera projection matrix P is written as 
[image]
Shouldn't the camera projection matrix P be 
[image]
Excuse me, is it something I ignored？

ishipachev, 2022-12-02:
@Zhang-Qinghui 
May you drop a snippet on this part of the paper?