AprilRobotics/apriltag #229: CMake Install Error

adarshrs, 2022-05-15:
Cloned this repository and ran the commands to install it using CMake from `~/apriltag`

Here is the result:
`$ cmake -B build -GNinja -DCMAKE_BUILD_TYPE=Release
CMake Error: The source directory "/home/adarsh/apriltag/build" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.`

All the files needed for configuration are actually in the main directory, and I was able to complete configuration by copying them into a new subdirectory called build.

But the following still produces an error:
 `$ cmake --build build --target install
Error: could not load cache`

Maybe I'm doing something wrong with the install, but if so, the instructions are a little unclear about what it might be

wxmerkt, 2022-05-15:
Hi @adarshrs 
This sounds like a configuration error. Can you configure you ran `mkdir build ; cd build ; cmake ..` to configure?

adarshrs, 2022-05-15:
That seemed to work, but I can't import apriltag into Python:
```
>>> from apriltag import apriltag
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: libapriltag.so.3: cannot open shared object file: No such file or directory

```

adarshrs, 2022-05-15:
Working after a reboot.

closed 2022-05-15 by adarshrs

christian-rauch, 2022-05-17:
`cmake -B build` should already create the `build` folder. Something like `mkdir build ; cd build ; cmake ..` should not be required.

adarshrs, 2022-05-17:
It doesn't seem to work that way - 

```
(base) adarsh@Atom:~$ git clone https://github.com/AprilRobotics/apriltag.git
Cloning into 'apriltag'...
remote: Enumerating objects: 840, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 840 (delta 0), reused 0 (delta 0), pack-reused 836
Receiving objects: 100% (840/840), 26.28 MiB | 24.29 MiB/s, done.
Resolving deltas: 100% (532/532), done.
(base) adarsh@Atom:~$ cd apriltag
(base) adarsh@Atom:~/apriltag$ cmake -B build -DCMAKE_BUILD_TYPE=Release
CMake Error: The source directory "/home/adarsh/apriltag/build" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
```

christian-rauch, 2022-05-17:
Not sure why it does not work for you. For me, `cmake -B build` creates the folder and configures the project:
```
christian@gungnir:~$ cd /tmp/
christian@gungnir:/tmp$ git clone https://github.com/AprilRobotics/apriltag.git
Cloning into 'apriltag'...
remote: Enumerating objects: 840, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 840 (delta 0), reused 0 (delta 0), pack-reused 836
Receiving objects: 100% (840/840), 26.28 MiB | 4.49 MiB/s, done.
Resolving deltas: 100% (532/532), done.
christian@gungnir:/tmp$ cd apriltag
christian@gungnir:/tmp/apriltag$ cmake -B build
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'Release' as none was specified.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.9.so (found version "3.9.12") 
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/apriltag/build
```

Are you maybe using an older version? I am using 3.16.3:
```
$ cmake --version
cmake version 3.16.3
```

adarshrs, 2022-05-17:
I'm on Ubuntu 18.04, with cmake 3.10.2
That's the newest version on 18.04

adarshrs, 2022-05-17:
I'm using 18.04 for ROS Melodic, as, I think, a lot of people still are. Maybe it would be helpful to add a note in the readme

robin-bird-go, 2023-08-26:
Hello there, I'm also a user of  ubuntu 18.04, and I have the same problem as you, and I tried following  steps and finally fixed  it, hope it could help
(I'm a Chinese student so my system language is Chinese  (,,・ω・,,))

```shell
  robinbird@robinbirds-ubuntu:~/Desktop/camera/april_tag_lib$ git clone https://github.com/AprilRobotics/apriltag.git
  正克隆到 'apriltag'...
  remote: Enumerating objects: 987, done.
  remote: Counting objects: 100% (411/411), done.
  remote: Compressing objects: 100% (113/113), done.
  remote: Total 987 (delta 339), reused 315 (delta 298), pack-reused 576
  接收对象中: 100% (987/987), 26.34 MiB | 3.01 MiB/s, 完成.
  处理 delta 中: 100% (620/620), 完成.
  robinbird@robinbirds-ubuntu:~/Desktop/camera/april_tag_lib$ cd apriltag/
  robinbird@robinbirds-ubuntu:~/Desktop/camera/april_tag_lib/apriltag$ mkdir buildrobinbird@robinbirds-ubuntu:~/Desktop/camera/april_tag_lib/apriltag$ cd  build/
  robinbird@robinbirds-ubuntu:~/Desktop/camera/april_tag_lib/apriltag/build$ cmake ..  -DCMAKE_BUILD_TYPE=Release
  -- The C compiler identification is GNU 7.5.0
  -- The CXX compiler identification is GNU 7.5.0
  -- Check for working C compiler: /usr/bin/cc
  -- Check for working C compiler: /usr/bin/cc -- works
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Check for working CXX compiler: /usr/bin/c++
  -- Check for working CXX compiler: /usr/bin/c++ -- works
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Looking for pthread.h
  -- Looking for pthread.h - found
  -- Looking for pthread_create
  -- Looking for pthread_create - not found
  -- Looking for pthread_create in pthreads
  -- Looking for pthread_create in pthreads - not found
  -- Looking for pthread_create in pthread
  -- Looking for pthread_create in pthread - found
  -- Found Threads: TRUE  
  -- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found version "3.6.9") 
  -- Configuring done
  -- Generating done
  -- Build files have been written to: /home/robinbird/Desktop/camera/april_tag_lib/apriltag/build
  ```