AprilRobotics/apriltag #231: Installation build error: "gcc: error: libpython3.10d.a: No such file or directory"

2shaar2059, 2022-05-31:
BLAH_BLAH_BLAH$ git clone https://github.com/AprilRobotics/apriltag.git
Cloning into 'apriltag'...
remote: Enumerating objects: 855, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 855 (delta 3), reused 6 (delta 1), pack-reused 840
Receiving objects: 100% (855/855), 26.31 MiB | 2.55 MiB/s, done.
Resolving deltas: 100% (539/539), done.
BLAH_BLAH_BLAH$ cd apriltag/
BLAH_BLAH_BLAH$ cmake -B build -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
-- Found CUDA: /usr/local/cuda-11.5 (found suitable exact version "11.5") 
-- Configuring done
-- Generating done
-- Build files have been written to: BLAH_BLAH_BLAH/apriltag/build
BLAH_BLAH_BLAH$ cmake --build build --target install
Scanning dependencies of target apriltag
[  2%] Building C object CMakeFiles/apriltag.dir/apriltag.c.o
[  5%] Building C object CMakeFiles/apriltag.dir/apriltag_pose.c.o
[  7%] Building C object CMakeFiles/apriltag.dir/apriltag_quad_thresh.c.o
[ 10%] Building C object CMakeFiles/apriltag.dir/common/g2d.c.o
[ 12%] Building C object CMakeFiles/apriltag.dir/common/getopt.c.o
[ 15%] Building C object CMakeFiles/apriltag.dir/common/homography.c.o
[ 17%] Building C object CMakeFiles/apriltag.dir/common/image_u8.c.o
[ 20%] Building C object CMakeFiles/apriltag.dir/common/image_u8x3.c.o
[ 22%] Building C object CMakeFiles/apriltag.dir/common/image_u8x4.c.o
[ 25%] Building C object CMakeFiles/apriltag.dir/common/matd.c.o
[ 27%] Building C object CMakeFiles/apriltag.dir/common/pam.c.o
[ 30%] Building C object CMakeFiles/apriltag.dir/common/pjpeg-idct.c.o
[ 32%] Building C object CMakeFiles/apriltag.dir/common/pjpeg.c.o
[ 35%] Building C object CMakeFiles/apriltag.dir/common/pnm.c.o
[ 37%] Building C object CMakeFiles/apriltag.dir/common/string_util.c.o
[ 40%] Building C object CMakeFiles/apriltag.dir/common/svd22.c.o
[ 42%] Building C object CMakeFiles/apriltag.dir/common/time_util.c.o
[ 45%] Building C object CMakeFiles/apriltag.dir/common/unionfind.c.o
[ 47%] Building C object CMakeFiles/apriltag.dir/common/workerpool.c.o
[ 50%] Building C object CMakeFiles/apriltag.dir/common/zarray.c.o
[ 52%] Building C object CMakeFiles/apriltag.dir/common/zhash.c.o
[ 55%] Building C object CMakeFiles/apriltag.dir/common/zmaxheap.c.o
[ 57%] Building C object CMakeFiles/apriltag.dir/tag16h5.c.o
[ 60%] Building C object CMakeFiles/apriltag.dir/tag25h9.c.o
[ 62%] Building C object CMakeFiles/apriltag.dir/tag36h10.c.o
[ 65%] Building C object CMakeFiles/apriltag.dir/tag36h11.c.o
[ 67%] Building C object CMakeFiles/apriltag.dir/tagCircle21h7.c.o
[ 70%] Building C object CMakeFiles/apriltag.dir/tagCircle49h12.c.o
[ 72%] Building C object CMakeFiles/apriltag.dir/tagCustom48h12.c.o
[ 75%] Building C object CMakeFiles/apriltag.dir/tagStandard41h12.c.o
[ 77%] Building C object CMakeFiles/apriltag.dir/tagStandard52h13.c.o
[ 80%] Linking C shared library libapriltag.so
[ 80%] Built target apriltag
Scanning dependencies of target opencv_demo
[ 82%] Building CXX object CMakeFiles/opencv_demo.dir/example/opencv_demo.cc.o
[ 85%] Linking CXX executable opencv_demo
[ 85%] Built target opencv_demo
Scanning dependencies of target apriltag_demo
[ 87%] Building C object CMakeFiles/apriltag_demo.dir/example/apriltag_demo.c.o
[ 90%] Linking C executable apriltag_demo
[ 90%] Built target apriltag_demo
Scanning dependencies of target apriltag_python
[ 92%] Generating apriltag_py_type.docstring.h
[ 95%] Generating apriltag_detect.docstring.h
[ 97%] Generating apriltag_pywrap.o
[100%] Generating apriltag.cpython-310d-x86_64-linux-gnu.so
gcc: error: libpython3.10d.a: No such file or directory
make[2]: *** [CMakeFiles/apriltag_python.dir/build.make:62: apriltag.cpython-310d-x86_64-linux-gnu.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:136: CMakeFiles/apriltag_python.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

christian-rauch, 2022-05-31:
Usually, you will need to provide a bit more information about your system (OS, version, ...) in order to get help.

2shaar2059, 2022-05-31:
Sure, does this help?

OS: Ubuntu 20.04.4 LTS x86_64
Kernel: 5.4.0-113-generic
CPU: Intel i7-6500U (4) @ 3.100GHz
GPU: NVIDIA GeForce GTX 950M
GPU: Intel Skylake GT2 [HD Graphics 520]

python3 --version: Python 3.10.4

gcc --version: gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0

christian-rauch, 2022-06-01:
Thanks. I am using the same system without issues. But my python version is:
```
$ python3 --version
Python 3.8.10
```

According to https://packages.ubuntu.com/focal/python3, the newest python3 package should be version `3.8.2-0ubuntu2`.

Are you using python3 from another source? My bet is that you have different versions of runtimes (`python3`) and development packages (`python3-dev`) installed. Both versions should match.

Did you try with the standard packages (version `3.8.2-0ubuntu2`) in Ubuntu?

closed 2022-06-03 by christian-rauch