AprilRobotics/apriltag #235: Some questions about reliability of identification

meihe323, 2022-06-16:
Im working on a project that landing a copter on the apriltags ,here comes some diffculties,the first is that when the copter is very close to the apriltag, I found that the shadow of copter covered some black blocks of tag, and I can't get good identification result.How can I do to deal with shadows to get better reliability of identification?

brmarkus, 2022-06-16:
Would you have a chance to add diffuse lights/LEDs to the bottom of the copter? Or add multiple tags next to each other? Or add a circle in a thicker line around it then you can switch the algorithm from Apriltag-detection to tracking the circle's center.

chutsu, 2022-10-27:
Very late to the party, but way back when I was doing a similar project; landing a quadcopter onto an AprilTag in outdoors. We basically used the "illumination invariant transform" proposed by Will Maddern:

Source:
> Maddern, Will, et al. "Illumination invariant imaging: Applications in robust vision-based localisation, mapping and classification for autonomous vehicles." Proceedings of the Visual Place Recognition in Changing Environments Workshop, IEEE International Conference on Robotics and Automation (ICRA), Hong Kong, China. Vol. 2. No. 3. 2014.

Link: https://www.robots.ox.ac.uk/~mobile/Papers/2014ICRA_maddern.pdf

The only requirement is the camera has to be RGB, and by manipulating the RGB channels via the transform you can form a 1 channel illumination invariant image. The result is if you print a color AprilTag you can mitigate the affects of shadows outdoors like so:

[image]

[https://www.youtube.com/watch?v=CgcqxZ-QGTc](url)