AprilRobotics/apriltag #238: Anyway to draw tag's coordinate frame in image?

liuchangji, 2022-06-25:
Thank you for sharing this excellent project. I wonder if there is any way to draw the coordinate axis of tag on the image? If not, I can only choose to use tag pose and camera intrinsics  to project axis to draw the tag's coordinate frame.

christian-rauch, 2022-06-25:
`apriltag_detector_detect` already provides the centre and corner coordinates in the `apriltag_detection` struct: https://github.com/AprilRobotics/apriltag/blob/v3.2.0/apriltag.h#L225-L230.

Just use this to draw the position and orientation of the frame. You do not need the camera intrinsics.

liuchangji, 2022-06-26:
@christian-rauch Thanks for your reply.Oh,I see it but I want to show axis in every frame.So I calculate it and project aixs's end point into pixel coordinate.This make something easier to understand for me.And I hope this function can be added into new version.
[image]

My code is below, you can try it in opencv_demo.cc :

 ```
while (true) {
        cap >> frame;
        cvtColor(frame, gray, COLOR_BGR2GRAY);

        // Make an image_u8_t header for the Mat data
        image_u8_t im = { .width = gray.cols,
            .height = gray.rows,
            .stride = gray.cols,
            .buf = gray.data
        };

        zarray_t *detections = apriltag_detector_detect(td, &im);

        // Draw detection outlines
        for (int i = 0; i < zarray_size(detections); i++) {
            apriltag_detection_t *det;
            zarray_get(detections, i, &det);
            info.det = det;
            double err = estimate_tag_pose(&info, &pose);

            // Pose To SE(3),but this is not necessary.
            Eigen::Matrix4d aprilTagRtMat;
            aprilTagRtMat(0,0) = pose.R->data[0];
            aprilTagRtMat(0,1) = pose.R->data[1];
            aprilTagRtMat(0,2) = pose.R->data[2];
            aprilTagRtMat(0,3) = pose.t->data[0];

            aprilTagRtMat(1,0) = pose.R->data[3];
            aprilTagRtMat(1,1) = pose.R->data[4];
            aprilTagRtMat(1,2) = pose.R->data[5];
            aprilTagRtMat(1,3) = pose.t->data[1];

            aprilTagRtMat(2,0) = pose.R->data[6];
            aprilTagRtMat(2,1) = pose.R->data[7];
            aprilTagRtMat(2,2) = pose.R->data[8];
            aprilTagRtMat(2,3) = pose.t->data[2];

            aprilTagRtMat(3,0) = 0;
            aprilTagRtMat(3,1) = 0;
            aprilTagRtMat(3,2) = 0;
            aprilTagRtMat(3,3) = 1;
            cout << aprilTagRtMat << endl;

            int sizeScale =1;//axis scale.

            for(int k = 2;k>-1;k--){

                line(frame, Point(det->p[0][0], det->p[0][1]),
                         Point(det->p[1][0], det->p[1][1]),
                         Scalar(0, 0xff, 0), 1);
                line(frame, Point(det->p[0][0], det->p[0][1]),
                         Point(det->p[3][0], det->p[3][1]),
                         Scalar(0, 0, 0xff), 1);
                line(frame, Point(det->p[1][0], det->p[1][1]),
                         Point(det->p[2][0], det->p[2][1]),
                         Scalar(0xff, 0, 0), 1);
                line(frame, Point(det->p[2][0], det->p[2][1]),
                         Point(det->p[3][0], det->p[3][1]),
                         Scalar(0xff, 0, 0), 1);

                //calculate axis's end point in camera coordinate.
                double x_end = aprilTagRtMat(0,k)*info.tagsize*sizeScale+ aprilTagRtMat(0,3);
                double y_end = aprilTagRtMat(1,k)*info.tagsize*sizeScale+ aprilTagRtMat(1,3);
                double z_end = aprilTagRtMat(2,k)*info.tagsize*sizeScale+ aprilTagRtMat(2,3);
                //normalize
                x_end = x_end / z_end;
                y_end = y_end / z_end;
                // transform point to pixel coordinate,this needs camera intrinsics
                x_end = x_end * info.fx + info.cx;
                y_end = y_end * info.fy + info.cy;

                if(k==0){
                    //X axis
                    arrowedLine(frame, Point(det->c[0], det->c[1]),
                             Point(int(x_end), int(y_end)),
                             Scalar(0, 0, 255), 2);
                    putText(frame,"x", Point(int(x_end), int(y_end)),2,1.0,Scalar(0, 0, 255));
                }
                if(k==1){
                     //Y axis
                    arrowedLine(frame, Point(det->c[0], det->c[1]),
                             Point(int(x_end), int(y_end)),
                             Scalar(0, 255, 0), 2);
                    putText(frame,"y", Point(int(x_end), int(y_end)),2,1.0,Scalar(0, 255, 0));

                }

                if(k==2){
                    //Z axis
                    arrowedLine(frame, Point(det->c[0], det->c[1]),
                             Point(int(x_end), int(y_end)),
                             Scalar(255, 0, 0), 2);
                    putText(frame,"z", Point(int(x_end), int(y_end)),2,1.0,Scalar(255, 0, 0));

                }

            }

            stringstream ss;
            ss << det->id;
            String text = ss.str();
            int fontface = FONT_HERSHEY_SCRIPT_SIMPLEX;
            double fontscale = 1.0;
            int baseline;
            Size textsize = getTextSize(text, fontface, fontscale, 2,
                                            &baseline);
            putText(frame, text, Point(det->c[0]-textsize.width/2,
                                       det->c[1]+textsize.height/2),
                    fontface, fontscale, Scalar(0xff, 0x99, 0), 2);
        }
        apriltag_detections_destroy(detections);
        imshow("Tag Detections", frame);
        waitKey(1);
        if (waitKey(30) >= 0)
            break;

    }

```

christian-rauch, 2022-06-26:
> but I want to show axis in every frame

What do you mean by this?

Projection the 3D coordinate axes into the image is certainly possible using the camera intrinsics. But if you only want to show the axes in the image, you can skip the projection and use the 2D coordinates directly. They should perfectly align with the tag edges and centre.

liuchangji, 2022-06-26:
> but I want to show axis in every frame
Sorry,I expressed it very unclear. That means "I want to show X-Y-Z axis in every video frame".
And I haven't started going through your paper yet, so I don't know which edge is parallel to the x-axis and which is parallel to the y-axis. That's why I came up with this method of projection which is more inconvenient.

closed 2022-06-26 by liuchangji