AprilRobotics/apriltag #239: Windows build regression when cross compiling

VincentMarnier, 2022-06-27:
**Describe the bug**
I am not able to build for windows anymore

**To Reproduce**
I use a docker image to build:
```
FROM alpine:3.15

RUN apk add cmake mingw-w64-gcc libpthread-stubs gcc make libtool musl-dev g++ gcompat
```
Though this should be the same regardless of your linux env
Then I run:
```
cmake -DCMAKE_SYSTEM_NAME=Windows -DCMAKE_C_COMPILER=x86_64-w64-mingw32-gcc -DCMAKE_CXX_COMPILER=x86_64-w64-mingw32-g++ -DBUILD_SHARED_LIBS=ON -DCMAKE_C_COMPILER_WORKS=1 -DCMAKE_CXX_COMPILER_WORKS=1 -DCMAKE_BUILD_TYPE=Release /apriltag
cmake --build . --config Release
```
It used to work. It does not after 2 changes, both specific to support windows build:

The first one is in `common/debug_print.h`:
```
#ifdef _WIN32
#define debug_print(fmt, ...) \
        do { if (DEBUG) fprintf(stderr, "%s:%d:%s(): " fmt, strrchr("\\"__FILE__,'\\')+1, \
                                __LINE__, __func__, __VA_ARGS__); fflush(stderr);} while (0)
#else
```
It works if I remove the whole ifdef _WIN32 and use the else content

The second one is linked to `common/pthreads_cross.h`:
The function `unsigned int timespec_to_ms(const struct timespec *abstime);` is now extern, thought it is defined static in `common/pthread_cross.cpp`:
```
static unsigned int timespec_to_ms(const struct timespec *abstime)
```

I don't really feel like making a PR as I am guessing those changes have probably been tested and maybe are probably necessary for some other projects. Maybe it is intended to be compiled directly on a windows system and not using cross compilation

**Expected behavior**
The windows dll is generated

** Input Image**
N/A

**Screenshots**
N/A

**Operating Sytem**
Alpine linux

** Installation Method**
None, I'm actually trying to build

** Code version **
Head is master

**Additional context**
None

christian-rauch, 2022-06-27:
Windows support changed with https://github.com/AprilRobotics/apriltag/pull/234 to remove the dependency on external packages. This works natively on Windows as you can see in the [last CI status](https://github.com/AprilRobotics/apriltag/actions/runs/2510288327). It is very likely that cross-compilation broke with this PR since the native build uses MSVC instead of MinGW.

We don't have a cross-compilation configuration for Windows. Given that there apparently are users that cross-compile for Windows on Lunux, it would make sense to fix the remaining issues and add a cross-compilation configuration to the CI.

@VincentMarnier Would you be able to send a PR with your changes that fix cross-compilation and maybe also add a cross-compile CI configuration?

VincentMarnier, 2022-06-27:
Hello,
I tried to compile with MSVC and the compilation work with and without my changes.
Thought in both cases I fail to use the produced dll.
I am actualy using the dll as a Unity3D Native plugin (https://docs.unity3d.com/Manual/NativePlugins.html) with a self made wrapper. It seems to work with the dll generated when cross compilling the same source code. I might be doing something wrong here; to be fair I aint comfortable pushing some code I could not test in your expected environment.
I might share a git patch or something and someone could test it?

christian-rauch, 2022-06-27:
> I am actualy using the dll as a Unity3D Native plugin

Out of curiosity, if you only need this for Unity, why don't you compile this plugin on Linux? As far as I know, you can use Unity also on Linux. Alternatively, if you use Unity on Windows, why don't you just compile it natively there?

> I might be doing something wrong here; to be fair I aint comfortable pushing some code I could not test in your expected environment.

Don't worry, you cannot push code to the main repository anyway. You have to send a pull request (PR) first.

> I might share a git patch or something and someone could test it?

The idea would be to send a pull request of now of your branches with those fixes to this project. People can then review this and we can test with the CI pipeline that this compiles.

christian-rauch, 2022-07-04:
@VincentMarnier Can you provide a bit more details in which situations you encounter this problem? Is this only for cross-building for Windows on Linux? If this is a valid use case, it would make sense to have this fixed upstream in this repo. Can you send a PR fixing your issue?

VincentMarnier, 2022-07-04:
Sorry I have been busy lately
Yes I encountered this problem only in this situation 
I will try to elaborate and make a PR Friday; I wont have access to my computer until then

closed 2022-07-11 by christian-rauch