AprilRobotics/apriltag #243: 'refine_edges' is an invalid keyword argument

DanielChaseButterfield, 2022-07-13:
**Describe the Bug:**
The file `apriltag_py_type.docstring` states the following:

```
The constructor takes a number of arguments:
...
- refine_edges: When non-zero, the edges of the each quad are adjusted to "snap
  to" strong gradients nearby. This is useful when decimation is employed, as it
  can increase the quality of the initial quad estimate substantially. Generally
  recommended to be on. Very computationally inexpensive. Option is ignored if
  decimate == 1. Default is True
```

However, when trying to use this argument, the following bug occurs:
```
TypeError: 'refine_edges' is an invalid keyword argument for this function
```

**To Reproduce**
Try to run the following command in a python file:
```
at_detector = apriltag(family='tag36h11', threads=1, maxhamming=3, decimate=1.0,
                           blur=0.0, refine_edges=0, debug=0)
```

**Expected behavior**
No error, just that the refine_edges variable in the AprilTag detector is set to false.

**Operating Sytem**
WSL2 - Ubuntu 20.04.4

**Installation Method**
I built AprilTag from source following the instructions in the README

**Code version**
package.xml has:
```
<name>apriltag</name>
  <version>3.2.0</version>
```

christian-rauch, 2022-08-15:
Can you send a PR to the documentation or Python API?

DanielChaseButterfield, 2022-08-15:
@christian-rauch A PR to add the refine_edges functionality, or to remove the reference to "refine_edges" from the documentation? I can definitely do the former, but I'm not sure how easy it would be for me to do the later.

christian-rauch, 2022-08-15:
I haven't looked into the python bindings or its documentation :-) I don't know if `refine_edges` was there before or is supposed to be there. Maybe the parameter is just called different in the API? If there is no such parameter or one with equivalent functionality, then it's easier to remove it from the documentation.

DanielChaseButterfield, 2022-08-15:
I just checked with lib-dt-apriltags (https://github.com/duckietown/lib-dt-apriltags), and it looks like their python bindings include ```refine_edges``` functionality.

This makes me believe that refine_edges should be there, but the python port didn't fully implement it. I guess the solution to fix this would be to implement this functionality.

I'll look into whether I'd be able to fix this myself.

DanielChaseButterfield, 2022-08-15:
Okay, should be fixed.

christian-rauch, 2022-09-03:
Fixed by https://github.com/AprilRobotics/apriltag/pull/248.

@DanielChaseButterfield If you write `Fixes #243` in your PR, then this will close an associated issue automatically when the PR is merged.

closed 2022-09-03 by christian-rauch