AprilRobotics/apriltag #244: Compilation fails on Ubuntu 16.04 ROS kinetic

ninja-hu, 2022-07-19:
**Describe the bug**
Following error is seen when I try to compile

[ 32%] Building C object CMakeFiles/apriltag.dir/common/image_u8.c.o
[ 35%] Building C object CMakeFiles/apriltag.dir/common/image_u8x3.c.o
[ 37%] Building C object CMakeFiles/apriltag.dir/common/image_u8x4.c.o
[ 40%] Building C object CMakeFiles/apriltag.dir/common/workerpool.c.o
In file included from /home/locobot1/apriltag_ws/src/apriltag/common/workerpool.c:32:0:
/usr/include/stdio.h:314:6: error: unknown type name ‘_IO_cookie_io_functions_t’
      _IO_cookie_io_functions_t __io_funcs) __THROW __wur;
      ^
CMakeFiles/apriltag.dir/build.make:398: recipe for target 'CMakeFiles/apriltag.dir/common/workerpool.c.o' failed
make[2]: *** [CMakeFiles/apriltag.dir/common/workerpool.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/apriltag.dir/all' failed
make[1]: *** [CMakeFiles/apriltag.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'apriltag': 
  Command '['make', '-j1']' returned non-zero exit status 2

**Operating Sytem**
Ubuntu 16.04

** Installation Method**
Following the locobot auto installation script provided here : https://www.trossenrobotics.com/docs/interbotix_xslocobots/getting_started/user_guide.html

** Code version **
8c45cb387bcc5e4d08b04f01bbc9ba83aae93444

I had an older version of the repo which works fine. The HEAD for the older version:
76482356ff7ecb3f4895b3b600d533b2e65980d9

christian-rauch, 2022-07-19:
Ubuntu 16.04 is EOL and not supported anymore. The CI pipeline only tests the last two LTS versions (20.04 and 22.04). From the [CI status](https://github.com/AprilRobotics/apriltag/actions/runs/2651512800), you can see that those build without issues.

Support for 16.04 has to be contributed by someone else. I can review a PR if it also contains the [CI workflow for 16.04](https://github.com/AprilRobotics/apriltag/blob/master/.github/workflows/ros_workspace.yml).

wxmerkt, 2022-07-20:
@ninja-hu I've provided a fix in #245

PR #245 (2022-07-21): Add back Ubuntu 16.04 support - resolves #244
We need to define `_GNU_SOURCE` for the initially reported error to go away. 

Since Ubuntu 16.04 ships by default with OpenCV 2.4, we need further changes for renamed variables and moved modules (`TickMeter` was in `contrib` back in 2.4).

Resolves #244
commit d6b7d1c2 (2022-07-21): Merge pull request #245 from AprilRobotics/ubuntu-16.04-support

Add back Ubuntu 16.04 support - resolves #244
closed 2022-07-21 by wxmerkt