AprilRobotics/apriltag #247: Memory corruption inside gradient_clusters?

Windwoes, 2022-08-15:
I'm building AprilTag from source (commit bee5efb1cc263ec1459bca46193e3428220d52ba) using a custom CMake file and adding a couple extra files to allow access from the JNI.

The AprilTag source code itself is *unchanged*. I'm having an issue where often AprilTag will segfault, on both Windows and Linux. The JVM error log shows the fault happened in the `gradient_clusters` frame.

```
C  [libc.so.6+0x165ec5]
C  [libapriltag.so+0x39d5f]  gradient_clusters+0x3e1
C  [libapriltag.so+0x3a54e]  apriltag_quad_thresh+0x3e6
C  [libapriltag.so+0x2d6f9]  apriltag_detector_detect+0x3e7
```

I also sometimes get the message `double free or corruption (fasttop)`

Interestingly, it works fine on Android. So it seems like there may a memory bug inside `gradient_clusters`. (Either that or I have a very strange memory issue in my JNI code, but I'm simply pointing AprilTag at a greyscale image buffer, and when it doesn't crash, it runs fine)

christian-rauch, 2022-08-15:
Its hard to tell without a backtrace or a reproducible case with the default examples.

There have been some issues reported with ASAN (https://github.com/AprilRobotics/apriltag/pull/213). Maybe fixing those will also solve you issue? Otherwise, can you run your example with ASAN (AddressSanitizer) or Valgrind?

Windwoes, 2022-08-15:
I tried to run with ASAN, but it seems that ASAN isn't compatible with the JNI. I will try to find some way to figure out more specifically where the issue is.

Windwoes, 2022-08-20:
Update: the crashing on Linux was due to accidentally mixing two versions of the Java side and native side JNI code.
I am still having intermittent crashes on Windows which I have not yet been able to track down.

christian-rauch, 2022-08-27:
@Windwoes Is your issue resolved? Can we close this?

Windwoes, 2022-08-27:
Unfortunately I haven't had any more time to look into the windows issue yet, but I hopefully will soon.

closed 2022-09-11 by christian-rauch