AprilRobotics/apriltag #253: memory leak when use "estimate_tag_pose"

why-freedom, 2022-09-06:
**Describe the bug**
A clear and concise description of what the bug is.
when i use estimate_tag_pose, the memory grow up slowly,  I use opencv_demo.

**To Reproduce**
Steps to reproduce the behavior:
[e.g. Run apriltag_demo on the attached input image.]

**Expected behavior**
A clear and concise description of what you expected to happen.

** Input Image**
If applicable, please attach the input image that reproduces the problem.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating Sytem**
[e.g. Ubuntu, iOS, Windows]
Ubuntu18.04
** Installation Method**
[e.g. I built AprilTag from source following the instructions in the README]
source README, cmake .. make 
** Code version **
What does the version tag in package.xml say? Or, if you have downloaded the code from github right now, say current github HEAD.
use master
**Additional context**
Add any other context about the problem here.

christian-rauch, 2022-09-06:
Are you using the `opencv_demo` or your own code? Just from your code, I can tell that you have to free `info` and `pose`.

1. Can you run this with ASAN and valgrind and report back any memory issues?
2. Can you provide a complete example (code, build script, input data) to reproduce this?

why-freedom, 2022-09-07:
> Are you using the `opencv_demo` or your own code? Just from your code, I can tell that you have to free `info` and `pose`.
> 
> 1. Can you run this with ASAN and valgrind and report back any memory issues?
> 2. Can you provide a complete example (code, build script, input data) to reproduce this?

# Code

why-freedom, 2022-09-07:
I slove the problem, thank you!

closed 2022-09-07 by why-freedom

wxmerkt, 2022-09-07:
Thanks for the update - can you please confirm what the problem was?