AprilRobotics/apriltag #257: estimate_tag_pose giving segmentation fault

stark710, 2022-09-11:
`   double fx = (FOCAL_LENGTH / SENSOR_WIDTH) * IMG_WIDTH;
            double fy = (FOCAL_LENGTH / SENSOR_WIDTH) * IMG_WIDTH;
            double fx = (IMG_WIDTH * 0.5) / tan(FOV_HORIZONTAL * 0.5 * PI/180);
            double fy = (IMG_WIDTH * 0.5) / tan(FOV_HORIZONTAL * 0.5 * PI/180);
            apriltag_detection_info_t info;
            info.det = det;
            info.tagsize = TAG_SIZE;
            info.fx = fx;
            info.fy = fy;
            info.cx = 0;
            info.cy = 0;
            // // Then call estimate_tag_pose.
            apriltag_pose_t pose;
            double err = estimate_tag_pose(&info, &pose);`

I get a seg fault when I run this code. The error is probably at this line because the print statements before this get executed. 

` double scale = info->tagsize/2.0;
    matd_t* p[4] = {
        matd_create_data(3, 1, (double[]) {-scale, scale, 0}),
        matd_create_data(3, 1, (double[]) {scale, scale, 0}),
        matd_create_data(3, 1, (double[]) {scale, -scale, 0}),
        matd_create_data(3, 1, (double[]) {-scale, -scale, 0})};`

Been struggling with this for two days. Can someone please help me?

christian-rauch, 2022-09-11:
What does `gdb` say where this crashes?

Can you provide a complete example (code and build scripts), including the input data, that causes the crash?

Usually when reporting crashes, you either have to provide sufficient information to fix the issue or an easy way to reproduce the issue on our side. Otherwise, this becomes just wild guessing.

stark710, 2022-09-11:
I am running it on a robot that does not have gdb. I am sorry about the insufficient data. 
Can you just run the function with default parameters if any? Does it pass then? You can reproduce the issue by giving the tag size as 0.128 m, using tag36h11_create() function , tag36_11 type AprilTag and any camera intrinsics that you have. If it does pass, please post the example code of detections and usage of this function. That could be useful for me too. 
If this does not work and you need the exact data of my case, please let me know. Thank you for the prompt response.

christian-rauch, 2022-09-12:
> I am running it on a robot that does not have gdb.

Unless this is some kind of embedded system, you should be able to install gdb via apt or any other Linux package manager. Have you tried to run this with `ASAN` enabled (`cmake -DASAN=ON`)?

If running gdb on the robot is not an option, can you first try to reproduce this on your desktop/workstation?

> Can you just run the function with default parameters if any?

There is an example use of `estimate_tag_pose` in https://github.com/AprilRobotics/apriltag/issues/253 that I adapted a little bit. I can run this without crashes. Can you apply the patch (you have to rename the file and remove the ".txt"): [0001-estimate-pose.patch.txt](https://github.com/AprilRobotics/apriltag/files/9545629/0001-estimate-pose.patch.txt) and see if this works? If you can reproduce the issue, then please share a patch file here so we can also reproduce this and fix the issue.

stark710, 2022-09-13:
I was able to get it to work. I was making some mistakes in using the detections. And as mentioned in issue #253, there were some memory leaks. I resolved them. However, I did not do dynamic memory allocation as mentioned as that would cause more problems for me. Would you recommend that?

Also, the code works when the pose is >=0.33 m. Within that, it results in segmentation fault again. Is that a known issue?

stark710, 2022-09-13:
It also invoked this error - `apriltag_pose.c:440:fix_pose_ambiguities(): Error, more than one new minimum found.
`

> I was able to get it to work. I was making some mistakes in using the detections. And as mentioned in issue #253, there were some memory leaks. I resolved them. However, I did not do dynamic memory [...]

christian-rauch, 2022-09-13:
> And as mentioned in issue #253, there were some memory leaks. I resolved them.

Can you provide details, and maybe an example, about this memory leak? Where are those leaks caused and how did you resolve them? I couldn't reproduce leaks and AddressSanitizer (a.k.a. ASAN) did not report any leaks related to the library. Eventually, this issue was also closed as fixed.

> However, I did not do dynamic memory allocation as mentioned as that would cause more problems for me. Would you recommend that?

Are you talking about memory allocation on stack/heap?

> Also, the code works when the pose is >=0.33 m. Within that, it results in segmentation fault again. Is that a known issue?

No. Again, can you provide an example? What do you mean with ">=0.33 m"? Is this the tag size or the z-distance from the camera?

Unless you provide enough details to reproduce or directly identify the issue, we will not be able to fix it. Only the issues in the issue tracker are known issues. If you encounter a new problem, please report it with sufficient information.

christian-rauch, 2022-09-28:
I am closing this for now due to the lack of a reproducible example. If this happens again and you can provide an example, you can reopen this issue.

closed 2022-09-28 by christian-rauch