AprilRobotics/apriltag #258: Can't compile python version error ubuntu 22.04

albertarla, 2022-09-23:
I'm trying to compile the library as said in the README file with cmake. I have python 3.10.4 on my system running with ubuntu 22.04 and a pyenv enviornment with python 3.10.0. If I execute `which python` and `python --version` the output is that I'm using the pyenv one with python 3.10.0. Then when calling `cmake -B build -DCMAKE_BUILD_TYPE=Release` I get the following outpu:
```bash
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so (found version "3.10.4") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/albert/apriltag/build

```
As you can see it found my python system version 3.10.4.

Then when trying to compile it i get the follwing error: 
```bash
[  2%] Building C object CMakeFiles/apriltag.dir/apriltag.c.o
[  4%] Building C object CMakeFiles/apriltag.dir/apriltag_pose.c.o
[  7%] Building C object CMakeFiles/apriltag.dir/apriltag_quad_thresh.c.o
[  9%] Building C object CMakeFiles/apriltag.dir/common/g2d.c.o
[ 12%] Building C object CMakeFiles/apriltag.dir/common/getopt.c.o
[ 14%] Building C object CMakeFiles/apriltag.dir/common/homography.c.o
[ 17%] Building C object CMakeFiles/apriltag.dir/common/image_u8.c.o
[ 19%] Building C object CMakeFiles/apriltag.dir/common/image_u8x3.c.o
[ 21%] Building C object CMakeFiles/apriltag.dir/common/image_u8x4.c.o
[ 24%] Building C object CMakeFiles/apriltag.dir/common/matd.c.o
[ 26%] Building C object CMakeFiles/apriltag.dir/common/pam.c.o
[ 29%] Building C object CMakeFiles/apriltag.dir/common/pjpeg-idct.c.o
[ 31%] Building C object CMakeFiles/apriltag.dir/common/pjpeg.c.o
[ 34%] Building C object CMakeFiles/apriltag.dir/common/pnm.c.o
[ 36%] Building CXX object CMakeFiles/apriltag.dir/common/pthreads_cross.cpp.o
[ 39%] Building C object CMakeFiles/apriltag.dir/common/string_util.c.o
[ 41%] Building C object CMakeFiles/apriltag.dir/common/svd22.c.o
[ 43%] Building C object CMakeFiles/apriltag.dir/common/time_util.c.o
[ 46%] Building C object CMakeFiles/apriltag.dir/common/unionfind.c.o
[ 48%] Building C object CMakeFiles/apriltag.dir/common/workerpool.c.o
[ 51%] Building C object CMakeFiles/apriltag.dir/common/zarray.c.o
[ 53%] Building C object CMakeFiles/apriltag.dir/common/zhash.c.o
[ 56%] Building C object CMakeFiles/apriltag.dir/common/zmaxheap.c.o
[ 58%] Building C object CMakeFiles/apriltag.dir/tag16h5.c.o
[ 60%] Building C object CMakeFiles/apriltag.dir/tag25h9.c.o
[ 63%] Building C object CMakeFiles/apriltag.dir/tag36h10.c.o
[ 65%] Building C object CMakeFiles/apriltag.dir/tag36h11.c.o
[ 68%] Building C object CMakeFiles/apriltag.dir/tagCircle21h7.c.o
[ 70%] Building C object CMakeFiles/apriltag.dir/tagCircle49h12.c.o
[ 73%] Building C object CMakeFiles/apriltag.dir/tagCustom48h12.c.o
[ 75%] Building C object CMakeFiles/apriltag.dir/tagStandard41h12.c.o
[ 78%] Building C object CMakeFiles/apriltag.dir/tagStandard52h13.c.o
[ 80%] Linking CXX shared library libapriltag.so
[ 80%] Built target apriltag
[ 82%] Generating apriltag_py_type.docstring.h
[ 85%] Generating apriltag_detect.docstring.h
[ 87%] Generating apriltag_pywrap.o
[ 90%] Generating apriltag.cpython-310-x86_64-linux-gnu.so
/usr/bin/ld: cannot find libpython3.10.a: No such file or directory
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/apriltag_python.dir/build.make:74: apriltag.cpython-310-x86_64-linux-gnu.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:115: CMakeFiles/apriltag_python.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
```

How can I solve it or does anyone have installed the library succesfully on ubuntu 22.04? Thanks in advance.

christian-rauch, 2022-09-23:
The output is quite clear that the file `libpython3.10.a` is missing. You will find this file in package [`libpython3.10-dev`](https://packages.ubuntu.com/jammy/amd64/libpython3.10-dev/filelist).

albertarla, 2022-09-23:
Hi, thanks for the answer. Yes the file is present in my computer. Sorry for not clearing that out at first place  in the issue. Here the output of locate:
```bash
albert@albert:~$ locate libpython3.10.a
/home/albert/.pyenv/versions/3.10.0/lib/libpython3.10.a
/home/albert/.pyenv/versions/3.10.0/lib/python3.10/config-3.10-x86_64-linux-gnu/libpython3.10.a
/usr/lib/python3.10/config-3.10-x86_64-linux-gnu/libpython3.10.a
/usr/lib/x86_64-linux-gnu/libpython3.10.a
```
Also the output of python --version and which python:
```bash
albert@albert:~$ python --version
Python 3.10.0
albert@albert:~$ which python
/home/albert/.pyenv/shims/python
albert@albert:~$ 
```

christian-rauch, 2022-09-23:
Can you try this without a virtual environment? Since CMake recognises that Python is installed but then does not set the library path correctly to find `libpython3.10.a`, it may get confused by different paths.

albertarla, 2022-09-26:
After doing `sudo apt install libpython3.10-dev` and compiling without the virtual env it worked. Thanks a lot for the support.

closed 2022-09-26 by albertarla