AprilRobotics/apriltag #260: Running AprilTag in Google Colab

JosephTLockwood, 2022-09-27:
**Describe the bug**
I am attempting to get Apriltag working in Google Colab. I get the following error when trying to `import`.
`ImportError: libapriltag.so.3: cannot open shared object file: No such file or directory`

**To Reproduce**
The code to reproduce this error can be ran and seen [here](https://colab.research.google.com/drive/1ewt7qQJGi8cYjxi57ePe3lHMzaGHgPBt#scrollTo=ZNbSW8DQAowO)

**Expected behavior**
I have tried other alternatives. Previously I used this [pip package](https://pypi.org/project/apriltag/) which works well. However, this slowed down quite a bit the more tags that where in the frame. I also would like to have estimate_tag_pose. This then lead me to [dt-apriltags](https://github.com/duckietown/lib-dt-apriltags) however this was slower (without even getting the tag pose) and seems to no longer be supported with updates.

christian-rauch, 2022-09-27:
Can you provide more details on how the apriltag library was installed and how to reproduce this locally? Can you run a `locate libapriltag.so.3` to check if the file is present on the system? It might be that you are just using the wrong library path and that therefore the library cannot be found.

Otherwise, I don't see how this issue can be fixed in this repo.

JosephTLockwood, 2022-09-27:
Hi @christian-rauch thanks for the response. I was trying to run this on Google Colab as a test. I wanted to make sure I could get it working on a clean platform (one where I had not previously install other Apriltags) before I tired on my device. Because I was running into issues on Google Colab I figured I was doing something wrong or Apriltags just didn't work on Google Colab. 

I just tried again and got a different error. `
ImportError: cannot import name 'apriltag' from 'apriltag' (unknown location)`
Here is a link to the Colab this will be the easiest way for you to see the code I am running and the outputs I got https://colab.research.google.com/drive/1ewt7qQJGi8cYjxi57ePe3lHMzaGHgPBt#scrollTo=57TS0VPmjGtZ 
If Apriltag for some reason does not work on Google Colab that is fine. I can try it on my device tonight but wanted to make sure I was following the steps correctly first.

christian-rauch, 2022-09-27:
I haven't used Colab before, so I don't know how you would install custom software on it. 

I did a quick test with:
```sh
cmake -B build -DBUILD_PYTHON_WRAPPER=ON
cmake --build build
cd build/
LD_LIBRARY_PATH=. ipython3
```
and this allowed me to run `import apriltag` inside ipython.

Unless you can reproduce your issue locally in a minimal example, I would assume that this is somehow caused by your setup and not the library itself.

JosephTLockwood, 2022-09-28:
Thanks so much for you help. I got it working @christian-rauch. I ended up putting this on my device [Oak-D-Lite](https://shop.luxonis.com/products/oak-d-lite-1) but it was only running at around 10fps. dt-apriltag was running around 30fps and just the plain apriltag was around 40fps. Any idea on why it is slower? 
dt-apriltag
[image]
apriltag-(Compiled from C)
[image]

christian-rauch, 2022-09-28:
There could be many reasons for the different performances. Make sure that you compile both in release mode and use the same settings, such as decimate, that will hugely impact the runtime performance.

I assume that you managed to fix your original problem? Can this issue be closed?

closed 2022-09-28 by JosephTLockwood