AprilRobotics/apriltag #261: Discrepancy Between Generated Families and Built-In Families

bhjelstrom, 2022-10-05:
The included "classic" 16h5 family includes 30 tags.

When I generate a classic 16h5 family using apriltag-generation and  "java -cp april.jar april.tag.TagFamilyGenerator classic_8 5", I get a family with 25 tags (and the family is in fact named 16h5).

When I generate a standard 41h12 family with java -cp april.jar april.tag.TagFamilyGenerator standard_9 12, the generated "TagStandard41h12" has 1721 distinct tags. The included TagStandard41h12 family has 2115 distinct tags.

Trying to understand the cause of the discrepancy. Thanks.

mkrogius, 2022-10-05:
This happens because you are probably using a different version of Java
than was used originally, which has a different random number generator

christian-rauch, 2022-10-07:
@bhjelstrom If you are still facing issues, I suggest reporting them to https://github.com/AprilRobotics/apriltag-generation where they can be better tracked.

closed 2022-10-07 by christian-rauch