AprilRobotics/apriltag #262: Occlusion robustness

AlexReimann, 2022-10-07:
We did some occlusion tests of an `tagStandard41h12` tag and the results were actually terrible.

[image]

[image]

[image]

[image]

Are other tags / tag families 
more robust to occlusion? The original paper shows some highly occluded but still working tags:

https://april.eecs.umich.edu/media/pdfs/olson2011tags.pdf

[image]

Here is a bag of our test:

https://energyrobotics-my.sharepoint.com/:u:/g/personal/alexander_reimann_energy-robotics_com/EZvgFJxKYiNHp1mj_XxDT-sBZv6rWAcsGvo2LIOaHa3fFg?e=s2bD8t

The camera has some exposure switching turned on to cover more lightning conditions.
And we know it's kind of a crappy camera, but it's real world application after all.

christian-rauch, 2022-10-07:
Your data is quite low resolution. Did you try higher resolutions?

Other than that, I can only say that testing with a single tag family is not really exhaustive :-) Did you test other tag families? Did you try to reproduce the results from the paper?

s-trinh, 2022-10-07:
None of AprilTag 2, AprilTag 3 are robust to occlusion. Only the first version is.
This is stated in the [AprilTag 2 paper](https://april.eecs.umich.edu/media/pdfs/wang2016iros.pdf):

> The original AprilTag detector used image gradients to detect high-contrast edges. This has the advantage of being
> robust to shadows and variations in lighting over previous methods which used naive thresholding. Detection of partially occluded tags was made possible by first fitting segments to the gradients, then searching over combinations
> of segments that formed four-sided shapes, or quads. A disadvantage of the segment-first approach is the large number of candidate quads that are generated.

> The improvements to the tag detector were inspired by user feedback about common use cases. We learned that in
> most deployments, detection of partially occluded tags is of limited utility. Occluded tags often have one or more bit
> errors, and most users disable decoding of tags with bit errors due to the impact on false positive rates. No known users
> accept tags with more than two bit errors, which enables a faster decode algorithm. In our experience, the increased
> detection speed is a favorable tradeoff against the ability to recover partially occluded tag borders.

christian-rauch, 2022-10-12:
@AlexReimann I am going to close this. If you need occlusion robustness, you can switch to an earlier version of the library.

AlexReimann, 2022-10-22:
Thank you very much for the info :+1: 

I actually tested with a different camera and the results there were a bit better:

[image]
[image]
[image]

GuidoBartoli, 2024-05-16:
> @AlexReimann I am going to close this. If you need occlusion robustness, you can switch to an earlier version of the library.

We are evaluating AprilTags for outdoor robot localization, can you recommend the most occlusion-resilient variant? (Tag36h11, TagStandard, TagCircle, ...)

jobermark-nrl, 2025-03-13:
> > [@AlexReimann](https://github.com/AlexReimann) I am going to close this. If you need occlusion robustness, you can switch to an earlier version of the library.
> 

The earliest tags in this repository appear to be AprilTag 3.  Where would someone find the earlier version 1 of this library that is more occlusion-tolerant ?  I checked  this site :https://april.eecs.umich.edu/software/apriltag  but the software directs to this repo.